Ultimaker/CuraEngine #154: Support for corner brims

minad, 2015-01-27:
I would like the possibility to generate corner brims (additionally to the normal brims) automatically. Would you be interested in such a feature?

[image]

Dim3nsioneer, 2015-01-28:
Interesting. What's the actual gain? 'Just' less print time or even an improved anti-warping-behavior?

BagelOrb, 2015-01-28:
Also a bit less post-processing of the model. (removing just the elephant feet instead of the whole brim)

Note that for organic shapes the computation time would go up, while the resulting brim would be nearly similar, given that organic shapes have many (outer) corners.

Moreover, we wouldn't like the GUI to be flooded with options which can come in handy only for a handful of cases.

I don't think the benefit exceeds the trouble of implementing it, but you are free to code it yourself and submit a pull request.
It would then probably be included as a 'hidden option' in a later version of Cura.

Ghostkeeper, 2015-01-28:
I don't think it would hurt to just add an additional entry in the "Platform Adhesion Type" dropdown. If it provides better or equally good anti-warping-behaviour, compared to the current brim, then I think it'd be good to add.

You'd have to define well what constitutes a corner though. You'd also want to apply these brims to rounded corners, so it's not just vertices with incoming and outgoing edges under an acute angle or something. It could involve some complicated geometrical algorithms.

BagelOrb, 2015-01-28:
Indeed it can become quite difficult, but I don't think it will provide better anti-warping behaviour.
I don't think there is a lot to gain from this idea, since it is basically a brim, but not at all locations.

minad, 2015-01-28:
I would suggest combining brim + elephant brim. For the model above this significantly improved adhesion since the corners are under much more stress than the rest.

However you can always add this brims manually. But then the paths are not nicely aligned. I want to have the brims going in circles around the model and the model filled by parallel lines. If you add one layer to the model with the brim (as above), the cura brim will go around but the elephant brim will be filled with parallel lines. Therefore it would be nice to have this in the engine.

minad, 2015-01-28:
A crude illustration :grinning: 

[image]

BagelOrb, 2015-01-28:
I think you misunderstand what are called elephant feat. 
They refer to the effect that when you print without any brim at all, the outer parameter of the first layer is spread out more, causing a kind of mini-brim.

I understand what you want, but the question is why do you want it?
Why not just use normal brim?

minad, 2015-01-28:
Ok, corrected it (Sorry, I misused the term).

As I wrote above, I want to have this because for models of that type the corners are under much more stress. If I use a normal brim (not too thick brim) the model above still detached at the corners.

Another possibility would be to implement some kind of "more focused" brim at parts which are under heavy stress. It would be great if CuraEngine could generate that automatically. This could be added as option "adaptive brim". As a heuristic one could use the angle of corners and make the brim thicker depending on the angle. What do you think?

minad, 2015-01-28:
@Ghostkeeper one could use the curvature of the edge as a measure if we have a corner.

BagelOrb, 2015-01-28:
If the corners don't get detached with 'corner brims' with a radius of, say 5, then they certainly also won't get detached when using a normal brim with width 5.

Do you want to exclude the parts of the normal brim which are not part of the corner brim just because it saves time?

minad, 2015-01-28:
@BagelOrb Yes you are right, it saves time and is more elegant. In the example above, the brim must have maybe about 30 lines to give an equally good result compared to the corner brims.

I see two solutions:

1. The brim width corresponds to the brim area and we add some kind of factor which measures how much the brim is redistributed to the corners. The algorithm should then reduce the brim width at the sides and increase the width at the corners.
2. Add an additional parameter "number of corner lines" which get somehow added to the corner brim.

However in both cases one might need additional parameters to somehow "tune" the detection of corners. In general I would prefer not to have to many options and just generate something reasonable by default.

It seems you are hesitant about implementing such a thing, which I understand when I take a look at https://github.com/Ultimaker/CuraEngine/blob/master/src/skirt.cpp which is kept very simple as of now. But maybe you still want to keep this issue open for discussion :)

daid, 2015-01-28:
Detection isn't actually the hardest part, it's generating the proper lines for those rounded parts, and print them properly. (There are quite a few choices to make there)

What I like about the current brim is that it usually peels off as a single thing. Making actually quite easy to remove.

minad, 2015-01-28:
@daid I think you already have everything it needs in CuraEngine. For example for an adaptive-width brim I would propose the following:

1. Print at first a few outer lines (like normal brim)
2. Print a few inner lines directly around the object (like normal brim)
3. Fill the gaps with the normal area filling algorithm

minad, 2015-01-28:
Then it will also peel off as easily as now.

BagelOrb, 2015-01-28:
another algorithm:
-for each 'outer corner' : generate a circle (or low fi : regular polygon)
-union all circle areas
-subtract the area of the first layer of the model
-perform iterative insets in order to fill in the resulting area, resulting in the corner brim

The problem is how to detect the corner points?
We cannot rely on the angle, since a corner might consist of several small line segments with each a small angle, while the resulting total angle of the corner is high.

daid, 2015-01-28:
@minad That's one of the ways to fill it up. I talked with BagelOrb at the office, and we came up with a few filling ideas quite quick. However, I'm not convinced that any of the proposed methods provides a better "keeping corners down" result, which is what actually counts at the end.

minad, 2015-01-28:
@daid @BagelOrb The result will probably not differ if you print a brim with a lot of lines. However for large objects this is really some kind of overhead. I have seen that a some makers provide their models especially with these corner brims (as above). So there must be some reason. I thought it would maybe a good idea to do that automatically.

I printed this box model from above twice, once without corner brim and normal Cura brim (of maybe 10 lines). The corners even detached in this case. And the other time I printed only with the corner brims and everything worked fine. But this was on my reprap delta printer without heatbed.

I think we probably agree that it would not be worth to implement a solution with thousands of lines of code, but currently skirt generation is only 50 lines. So maybe there is room for improvement without introducing to much additional complexity.

closed 2015-01-29 by BagelOrb

minad, 2015-01-29:
Not worth considering?

minad, 2015-01-29:
Would you be interested if I provide a patch or is this out of the question?

daid, 2015-01-30:
I would be interested yes, if the code isn't too complex.

BagelOrb, 2015-01-30:
I closed the issue because I feel that the discussion is over. You can still implement it and present a pull request, which we will probably approve if the code is up to standard.

minad, 2015-01-30:
Ok :)