Ultimaker/CuraEngine #156: Minimum layer time problem with dual extruder

presslab-us, 2015-01-29:
When printing with dual extruders, it will print two layers at a time with each extruder.  This is to reduce switching extruders often, and also to work with the wipe tower.  But this also causes the minimum layer time to be violated because the enforced layer time is for the whole layer and not for each extruder on that layer.

BagelOrb, 2015-01-30:
Do you mean the minimal layer time is violated in between switching nozzle? Or that it is violated in between two layers with a single nozzle? Or is it violated in between layers for only one nozzle?

presslab-us, 2015-01-30:
It is violated between two layers with a single nozzle.  For example, say the layer has a lot of one material and not much of the other, this is the worst case. This is the sequence of events: print small part, increment layer, print small part, switch extruder, print large part, increment layer, print large part.  The small part does not have enough time to cool between layers.

closed 2016-10-10 by BagelOrb