Ultimaker/CuraEngine #159: Missing layer part holes

BagelOrb, 2015-02-23:
In the attached STL file, one of the nine holes in the top is missing.
The geometry is correct, and this can also be seen from the x-ray view (see below).
The behaviour is present in at least Cura 14.09 and 15.02.1.

It is no visualization issue.
x-ray) I have verified the geometry by hand and
toolpath) I have printed the object with the specific fault.

Could this be a clipperlib related issue?

[image]
https://www.youmagine.com/designs/missing-layerpart-holes

daid, 2015-02-23:
What happens if you disable "fix horrible type A"?

BagelOrb, 2015-02-24:
Indeed disabling the option [combine-everything type-A] was the culprit.

The problem is therefore with with that option.
Why would this option break correct geometry?

daid, 2015-02-24:
Because the normals of the inner cylinder are incorrect then (thus, an incorrect mesh)

BagelOrb, 2015-02-24:
Is there a way in Cura to see what the normals are?

closed 2015-04-14 by BagelOrb