Ultimaker/CuraEngine #160: Need an option to remove 100% "infill" adjacent to perimeter passes

dexter-day, 2015-02-25:
As shown below in a screenshot of the Lulzbot version of Cura, I'm wanting to remove this unnecessary 100% infill "wall" that generates adjacent to perimeter passes. This "wall" of full infill normally is used to support a angled perimeter when the perimeter is coming in towards the body of the part, but in the case shown below (circled in black) the perimeters are moving outward--overhanging over the printing surface--making the full infill unnecessary, time consuming, and hard on the mechanical components of the printer (quick back and forth motions). In the gray screenshot below, you will see the desired effect I'm looking for--taken from Makerbot Desktop. Makerbot Desktop only places "full" infill next to perimeter passes when the next pass in the 'z' axis will be placed on top of that "full" infill.
[image]

BagelOrb, 2015-02-26:
Isn't this caused by the bottom thickness? Because it is slightly angled, it counts as bottom.
Does the problem also occur when you reduce the Bottom / Top thickness?

dexter-day, 2015-02-26:
I tested changing the Bottom / Top thickness setting, and that did affect the area in question. Changing the bottom / top thickness to '0' completely removed the material I'm wanting to remove. But I really don't want to make the bottom / top thickness less than typical. Can a new variable be created to seperate the bottom / top thickness from the 'xy' direction thickness of the "bottom" layers created underneath overhangs between the actual top / bottom of the part--i.e. the body of the part?

BagelOrb, 2015-03-03:
It would be hard to separate the xy-thickness from the z-thickness. 

Furthermore I don't really see why we should. Suppose the bottom is ragged, such that it consists of small dramatically angled parts. According to your rule it would then get less than "full" material, while it should still support the whole mesh.

The top/bottom-thickness is used to control the thickness of the outer shell of the object. Having different outer shell thicknesses at different places, depending on the geometry of the object is outside the scope of the Cura project.

closed 2015-03-03 by BagelOrb

daid, 2015-03-04:
Even more, I say that Slic3r is doing the wrong thing there. As it's making sections thinner then you actually specified.

dexter-day, 2015-03-04:
Those sections should be controlled by the wall thickness settings, not the bottom / top thickness settings in my opinion...

BagelOrb, 2015-03-04:
They should be controlled by both, since a sloped wall can be viewed partly as bottom and partly as wall.

dexter-day, 2015-03-04:
But my bottom / top thickness is the same as my wall thickness. Shouldn't Cura simply realize that the wall thickness is enough to satisfy the strength requirements of the sloped wall bottom / wall layers?