Ultimaker/CuraEngine #162: Cura "crops" gcode file

Toffe87, 2015-03-04:
Pictures:

Bed: http://link.jepb.no/a2mb
Basic settings: http://link.jepb.no/a2lC
Advanced settings: http://link.jepb.no/a2xe
Expert settings: http://link.jepb.no/a2Pt

Files: 
2.2mb gcode file: http://link.jepb.no/a2Zk
12.1mb gcode file: http://link.jepb.no/a2sb
Ultimaker save file for the whole bed:  http://link.jepb.no/a2jA

When i select "Tools -> Print one at a time" i get a 2.2mb gcode file which stops at line 79318 (not containing the END code. Line looks like this:

G1 F3000 X

When i select the "Print all at once" i get approx the same time, meter and gram and when i export gcode its fine, it contains all the code + end code. The file now is 12.1MB

This outcome will be the same every time i try changing (tried a few times to see if it was one time occurence). I reloaded all objects, tried to move them equal to the picture and the error did not occur this time. One time error?

So the tips: Cura should check if the end code has properly been added to file so it can warn us if the file is bugged?

daid, 2015-03-04:
What does the engine log say? (it's in the file menu)

Toffe87, 2015-03-04:
Ah, have closed cura and as i said i could not reproduce the error. I found out that moving 1 of the objects just a little fixed it so it saved a 12mb file on both print all at once and print one by one. But since cura dosent save the position of the objects on the bed i cant reproduce it :/

closed 2015-03-04 by BagelOrb