Ultimaker/CuraEngine #164: Wall infill: Zig/zag pattern no lines

sselkirk, 2015-03-04:
Is it possible to configure CuraEngine to print straight lines for wall infills instead of zig/zag? I've played with the settings and cannot find.

Presently it prints dual lines for inner and outer perimeter (good), then uses a zig/zag for the infill (bad), consequently the hotend does quite a violent 'squiggling' motion on the infill putting considerable additional pressure on the hardware. Reducing the print speed to so the squiggling in less violent increase print time considerably.

A more efficient method would to infill with lines, as Slic3r does. However I don't want to use Slic3r because its terribly in efficient on travel more generally

The image shows slicing outcomes from CuraEngine and Slic3r on a 2mm wall. Green lines show hotend travel path.

Any help would be greatly appreciated

sselkirk, 2015-03-04:
[image]

BagelOrb, 2015-03-09:
The zigzag occurs because the path optimizer connects adjacent infill lines with the wrong ends. Instead of the nearest two ends of two adjacent infill lines, the end on the right of the first line is connected to the end on the left of the other.

There will soon be an option which prevents this!

sselkirk, 2015-03-09:
cheers thanks Tim @BagelOrb
Could you let me know when this option is available?

closed 2015-03-10 by BagelOrb