Ultimaker/CuraEngine #165: slicing small radii

alexanderswo, 2015-03-05:
HI together, I think there is some problem with slicing small radii (r=0,25mm / d=0,5mm).

I also posted this on: http://umforum.ultimaker.com/index.php?/topic/9884-printing-small-radii-problem/

Please take a look at this:
[image]

There seems to be some "defects" in the slicing when you look closer, at least when I print this, the printhead makes unregular moves following the path.

I hope this can be fixed somehow. If you need to know more details, let me know.

Rep Host V1.0.6
DeltaTower

daid, 2015-03-09:
Slicing looks exaclty as I would expect.

Print looks like your printer could use some more tweaking (not sure in which areas, little experience in deltas, but I'm 100% sure my Ultimaker wouldn't have issues with printing the above paths)

closed 2015-03-09 by daid

BagelOrb, 2015-03-10:
The print looks so bad because of the overlapping perimeters problem. This problem is being worked on.