Ultimaker/CuraEngine #168: JSON output mode?

peteruithoven, 2015-03-10:
It's currently challenging to work with CuraEngine's output. 
When starting a slice it outputs (over stderr), version and license info, but also errors on unsupported settings. Example:
```
Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Default config 'default.cfg' not used
Setting not found: extruderOffset [object Object],[object Object],[object Object],[object Object]
Unknown exception
```
This means we have to go through the lines and look for known error names. 
It would be great if the CuraEngine could communicate in json if a `-j /` `--json` parameter is supplied. 

This would also make it much easier to parse the progress feedback.

oliv3r, 2015-03-10:
Take a look at the PinkUnicorn editon of the CuraEngine. The current curaengine is on life support and won't see much use in a year from now.

peteruithoven, 2015-03-10:
Valid point, thanks.
If it includes JSON output and/or better parsable output this issue can be closed.

gsmith-to, 2015-04-23:
I'm a big fan of JSON - far better than XML when the data is a bunch of numbers - but an issue with JSON is that it is not, by design, stream-parseable. python's json, for instance, will only read the entire structure. So for data representing enormous Gcode files that might be considered problematic.
A way to get around this, which I've used before, is to break up the data into informal 'records', and write one per line ... fairly easy when you are are writing from C++, e.g.

{ "info": { "blah": [ "something"], "somevalue": 0, ... }, "grecords": [
{  "rectype": "tempset", "ext": 0, "val": 250},
{ "rectype": "toolpath", "ext": 0, .... },
{  ... more records }
]}

So it's a valid JSON file, but you can also choose to read it in line-by-line from python and convert each line as a single JSON object (the first one needs a "]}" at the end, and the trailing commas need deletion first, of course). Also, easy to use grep to extract interesting rectypes for inspection.
Other methods are possible, e.g. you could have 'records'  bracketed by lines that start with '{' and lines that start with '}'.

Ghostkeeper, 2015-04-23:
There seem to be plenty of Python modules that read JSON as a stream rather than completely storing it in memory. It's worth looking at something like [ijson](https://pypi.python.org/pypi/ijson/) if we are to read such large files. Implementing this should be no problem at all with Python's nice dependency framework.

The C++ side is a little more difficult though. I don't know how JSON is loaded at the moment. It would be a shame if we were to have to include additional dependencies for this.

gsmith-to, 2015-04-24:
Indeed, I stand corrected. Thanks for the link.

BagelOrb, 2015-08-27:
CuraEngine should communicate using the commandSocket.

closed 2015-08-27 by BagelOrb