Ultimaker/CuraEngine #169: Docs?

peteruithoven, 2015-03-10:
I found some info about the PinkUnicorn version on the forum: http://umforum.ultimaker.com/index.php?/topic/8606-whats-happening-in-cura-land/
But besides that information is scarce. I understand time is also scarce, maybe I'm missing a obvious resource. 
The forum post does mention documentation. 

I'm mostly interested in how to use the CuraEngine as slicer in another application, so information about settings, how to communicate with it etc.

BagelOrb, 2015-03-10:
Brief response: communication via libArcus.

Here's how I call it from the terminal:
./CuraEngine -v -s filamentFlow=100 -s layerThickness=100 -s initialLayerThickness=300 -s supportZDistanceBottom=150 -s supportZDistanceTop=150 -s supportBottomStairDistance=500 -s initialSpeedupLayers=4 -s minimalFeedrate=10 -s preSwitchExtruderCode= -s insetXSpeed=30 -s retractionZHop=0 -s extruderOffset3.X=0 -s extruderOffset3.Y=0 -s gcodeFlavor=GCODE_FLAVOR_ULTIGCODE -s postSwitchExtruderCode= -s retractionSpeed=40 -s infillOverlap=15 -s inset0Speed=20 -s coolHeadLift=0 -s extrusionWidth=400 -s upSkinCount=6 -s initialLayerSpeed=20 -s minimalLayerTime=5 -s infillSpeed=40 -s fanSpeedMax=100 -s enableCombing=1 -s fanSpeedMin=100 -s supportXYDistance=700 -s supportExtruder=-1 -s supportType=ZIGZAG -s supportZDistance=150 -s supportEverywhere=1 -s supportAngle=60 -s supportJoinDistance=600 -s supportBridgeBack=100 -s supportSpeed=20 -s supportConnectZigZags=1 -s supportAreaSmoothing=600 -s supportMinimalAreaSqrt=1000 -s supportTowerDiameter=2000 -s supportTowerRoofAngle=65 -s areaSupportPolyGenerator=1 -s filamentDiameter=1128 -s fanFullOnLayerNr=2 -s extruderOffset1.X=18000 -s extruderOffset1.Y=0 -s endCode=M25 -s minimalExtrusionBeforeRetraction=20 -s retractionMinimalDistance=1500 -s objectSink=0 -s retractionAmount=4500 -s skinSpeed=25 -s skinPattern=SKIN_LINES -s startCode=";FLAVOR:UltiGCode
;TIME:14228901
;MATERIAL:3058
;MATERIAL2:0" -s skirtDistance=0 -s skirtMinLength=150000 -s skirtLineCount=10 -s skirtSpeed=30 -s extruderOffset2.Y=0 -s extruderOffset2.X=0 -s fixHorrible=1 -s layer0extrusionWidth=400 -s moveSpeed=150 -s supportLineDistance=2666 -s retractionAmountExtruderSwitch=16500 -s sparseInfillLineDistance=2000 -s insetCount=2 -s downSkinCount=6 -s multiVolumeOverlap=150 -s position.X=115000 -s position.Y=112500 -s position.Z=0 -s infillPattern=INFILL_ZIGZAG -s extruderNr=0 -s avoidOverlappingPerimeters=1 -o "test.gcode" "Cura_test.stl"

paologlim, 2015-06-30:
@BagelOrb Do you know where we can get a full list of valid options for the -s parameter?

BagelOrb, 2015-07-01:
My comment above is outdated. Read the README of https://github.com/Ultimaker/CuraEngine
The settings can be found at https://github.com/Ultimaker/Cura/blob/master/resources/settings/fdmprinter.json

closed 2015-07-01 by BagelOrb