Ultimaker/CuraEngine #170: Feature Request: Continuous channel within infill to allow for easy filling of model with other materials

williammincy, 2015-03-14:
In order to get a greater strength or weight to a model the typical way to accomplish this is to increase the level of infill when printing. But there is another way to do this that conserves filament and can make for stronger models: Filling the model with another material such as sand or some other fluid that hardens.

This technique is not new and can be found on a few blogs but what is missing is a feature/option within Cura that gives a continuous channel within the model's infill to make later filling more balanced. The goal is to get as much material spread throughout the model as evenly as possible.

Having a continuous channel that makes its way through a print will allow a secondary material to flow into the print more reliably. It would also allow for the infill to be kept to a lower level but still be able to fill the print to increase its strength by adding another material to fill in the spaces within the model. It would also print much faster than a print having a higher infill.

I have been searching 3D applications and none that I've come across do this currently, and would be a killer feature for Cura, setting it even further apart from others that try to do what it does.

BagelOrb, 2015-03-16:
What do you specifically propose?

Do you propose to have an infill method which skips some layers at some points, thereby creating holes in the pattern for external material to flow through? That sounds like a hard job for limited functionality.

Of course you are free to implement it yourself and submit a pull request.

cscott, 2015-03-16:
I think you'd want to look at https://en.wikipedia.org/wiki/Space-filling_curve -- but the problem in its general form is very complicated.  You seem to want to guarantee that the interior spaces are always completely connected, no matter how small or oddly-shaped they are.  I'm not sure this is even possible.

I think you need to come up with a concrete algorithm before this is a bonafide feature request.

BagelOrb, 2015-03-17:
It is probably too hard to think of an algorithm which satisfies all the requirements, if it's even possible.

However, Hilbert lines certainly are an interesting infill pattern!
I don't know whether infill material could easily be injected, however.
Because of all the curves and cravices the injected material is blocked and air cannot easily escape.
Moreover, the print head would have to make a lot of corners, which slows down the print significantly. 

An easy alternative to Hilbert lines would be a simple zigzag line, which would also cover the area. I don't think Hilbert lines are the way to go here.

BagelOrb, 2015-04-14:
Another idea would be concentric infill and leave out one segment in each polygon; e.g. the rightmost segment.

cscott, 2015-04-14:
That's an interesting idea! Delete the rightmost segments in the concentric
inset until the segments you've removed have a cumulative length of <x>,
which is whatever you need for good infill flow. Perhaps alternate between
left side and right side (or even top and bottom) as you nest the shells
deeper so that one side doesn't end up weaker.  But that increases the
distance the infill must flow.  Perhaps alternating sides with z depth is
better.

BagelOrb, 2015-04-15:
I've never filled a print. How do you make the air escape?

You would have to drill a hole in the print to even start filling it, right? Couldn't you just drill further, also making holes in the concentric infill?

I think my time is better spent improving Cura in other ways.

However, if you submit a pull request, we can incorporate your code!

williammincy, 2015-08-27:
It's been a busy few months but I finally got around to solidly thinking about this and I have a few good ideas on how to make this work, though haven't figured out an automatic (programmatic) way to do this just yet. 

The simplest way to do this that I've come up with is to work with the existing grid but to alter the infill grid to try to make sure that there is ample opportunity for a liquid to find its way into as much of a print as possible. My initial idea is simply taking diamond patterns out from the infill grid so that each row and column put down doesn't form a segmented chamber. 

My initial writeup with some images showing what I am considering can be seen here: http://productiondojo.com/2015/08/26/feature-request-continuous-channels-within-3d-print-infills/

@BagelOrb - filling a print would ideally involve two holes, one for inflow of material and the other to let out air. You can do it with one but it requires more patience as air bubbles find their way out. It also seems like we were thinking along the same lines as some of the things you put out as possibilities I also hit on as well. 

I just forked the repo and found the file that defines the infill methods, so I'll will try playing with this a bit to see if I can generate the type of grid I'm looking for and will happily send a pull request if anything pans out.

BagelOrb, 2015-08-28:
A totally different idea is the following.

When filling a print afterwards you don't really need infill; you just need to support the top surface. Perhaps it's possible to 'abuse' the support system such that it generates support inside the print. That way not the whole mesh will be filled (generally) and the air around the inside support will make it possible to fill the air around the zigzag pattern.

I don't know if I will come around to this soon, but I think it is a legitimate feature request.

williammincy, 2015-08-28:
Thanks @BagelOrb, we really appreciate your time and the work done to make and maintain Cura. 

I think we are pretty much on the same page. I mentioned using existing support calculations that are used for external support but didn't expand on it as you have. That would be great but currently beyond my C programming skills.

But I may want this badly enough to get better at C, so I will try working the modified lattice for now and look into how external supports are calculated as a next topic to research.

BagelOrb, 2017-05-31:
With the newest Cura you can enable Hollow Out Object, which will cause the support system to put support inside the object. If you set the support pattern to ZigZag you have the highest chance of getting it all filled up.

We don't have functionality to introduce holes in the infill yet, though.

I have just thought of an infill pattern which has two continuous channels, but it is a concentric based pattern, so consectuive layers aren't guaranteed to sit on top of eachother.

On the other hand you could use Gradual Infill to minimize the amount of infill needed.
Most of the infill would then be reachable from the lower parts of your model.

worker72, 2017-06-01:
I tried "lines" infill type, with double infill layer thickness. 
I took a simple test object and sliced it for a 0.8mm nozzle. Printed the object with the 0.8mm nozzle with 0.2mm layerheight, and 0.4mm infill thickness (a setting that is hidden by default).
The results are nice, see pictures. The infill lines are stacked on each other, and there's a lot of air in between. 
In this case I used 20% infill.
Maybe that's already enough?

[image]
[image]
[image]