Ultimaker/CuraEngine #171: Unable to reset/move printer head when print is cancelled.

tebruno99, 2015-03-15:
Without manual control I'm unable to move the printer head so I can get the failed object off the bed.  I have to keep starting up RepeaterHost to move the head out of the way.

BagelOrb, 2015-03-16:
You can just move the print head by hand. Sometimes the motors block you from moving the head and platform. In such a case you can just turn the machine off and move all parts where you like.

You talk about manual control, which is an issue with the firmware, not with CuraEngine.

tebruno99, 2015-03-16:
How can no gui for sending commands to the printer be a firmware problem?

BagelOrb, 2015-03-16:
Aha. It wasn't clear to me you were talking about GUI controls.
This is not a problem with Ultimaker/CuraEngine, but with daid/Cura.

Perhaps direct control of the printer from the GUI will be worked on in the future; let's hope so.

cscott, 2015-03-16:
Those buttons are in the Pronterface UI, there's an option to select it.

Still: this is the wrong repository for this bug.

tebruno99, 2015-03-16:
My apologies, I thought CuraEngine/Cura was the same thing. I will look for the correct repository.

closed 2015-03-17 by BagelOrb