Ultimaker/CuraEngine #172: No support under pointy objects.

cscott, 2015-03-16:
The CuraEngine "Overhang angle for support" neglects to consider that the geometry in question might not be a continuation of a structure underneath it, but rather a pointy bottom.

For example, in http://www.thingiverse.com/thing:351620 there are several "pointy bottomed" areas, which don't get any support at all with Cura's default settings (overhang angle ~60 degrees).  If you reduce the overhang angle to 35 degrees you can get support generated, but then it also generates unnecessary support for several other areas.

CuraEngine should look not only at the inclination of the triangle, but also whether there is any fill in the layer underneath.

A related issue is that the generated fill often doesn't actually have any plastic underneath the pointy-bottomed area (see attached image).  In-filling the "top" area of the support would help a lot here (as was mentioned in issue #61 where unifying rafts and supports was proposed).
[image]

daid, 2015-03-16:
We're working on a new algorithm that fixes these issues.

closed 2015-03-16 by daid

cscott, 2015-03-16:
Cool, thanks!

It might be worth leaving this issue open until the new algorithm lands in order to reduce duplicate bug reports.  I searched through the first two pages of open bugs without finding anything similar before I took the time to file this one.