Ultimaker/CuraEngine #176: Change layer size at specified height

Cosmo3d, 2015-03-25:
Could be very useful to add a new function that changes the layer height to a specified level. For example, if you were to print a home, would it be possible to print at a thicker layer when you print the vertical walls and smaller print layer when you come to print the roof. This could decrease the print time whenever an object is made of vertical and sloped parts, without affecting the quality of the result in non-vertical sections. This process should be possible repeatedly during one printing.

daid, 2015-03-26:
Extremely complex, not going to happen anytime soon.

Renha, 2016-02-17:
hi @daid , could you please explain why is it so complex, if you'll have some time?

Ghostkeeper, 2016-02-17:
Let's say you're printing a house with a pointy roof, and so you will have a vertical wall towards the top of the roof right next to a slanted roof, at the same height. Will you print the vertical wall with a large layer height or does the entire layer need to be made smaller if there is a less steep face present? What if you're printing your house next to a car and the roof of the car is next to the vertical wall? Can it be printed with different layer heights?

And what about support? Does the support have to be sliced at the same layer height?

When mixing different layer heights, collision checking becomes a mess. For instance, for combing, you can't reliably trace around the walls with 5mm distance to avoid hitting the walls with an oozing nozzle. Because not everything that you could hit is on the z-coordinate you're currently slicing, you would have to take neighbouring layers into account as well.

Adapting layer height based on slant is also patented.

Renha, 2016-02-17:
@Ghostkeeper there was no multiple parts in original question, it is obvious that managing multiple layer heights on same actual layer is hell. I was wondering mostly on why is it complex to handle with changing layer size at specified height for whole print job. Of course with support, all extruders together, etc

BagelOrb, 2016-02-17:
I think a simple approach might be possible, but it would lead to unwanted behaviour. There will be sudden changes in layer height which causes a sudden change in required flow which causes there to be large blobs when switching to a smaller layer height. The alternative is to make the speed depend on the layer height, which is in itself quite hard, since the actual speeds don't always come near the requested speeds. (The speed settings are actually upper bounds to the speeds.)

Another complexity is that when checking the height of a layer based on the orientation of the faces in that layer, you will first need to know which faces are in the layer, for which you must know the exact position of the layer, for which you must know the layer height. This requires some difficult algorithm to eventually converge to the correct layer height.

Renha, 2016-02-20:
@BagelOrb agreed. I've non meant automatic layer height calculation, I think @Cosmo3d haven't too.

That feature could replace "first layer height", could help printing 5.5mm cube with accurate height and 0.2 mm layers (by making last layer 0.1 mm).

BagelOrb, 2016-03-09:
This should be done by a postprocessing plugin.

closed 2016-03-09 by BagelOrb