Ultimaker/CuraEngine #188: initialLayerFlow besides initialLayerThickness

BagelOrb, 2015-04-20:
Why adjust the height of the first layer?

I feel as though people abuse this setting to get the flow up, which means that after the bed levelling wizard, they move the platform up a bit so that the distance to the nozzle is .1 mm again, and the flow is effectively trippled. Such procedure would lead to the first 0.2 mm being cut off from the object.

I propose to set the standard initial layer thickness to 0.1m (or maybe 0.15mm) and set the initialLayerFlow to 200%

Dim3nsioneer, 2015-04-20:
Although I think that the 0.1mm difference in height is not crucial for most prints I agree. I also belong to those users who don't use any bed levelling wizard but level the bed manually during brim or skirt printing.

Renha, 2015-04-22:
> Why adjust the height of the first layer?
Because of stickness difference for some materials and surfaces. For example, nylon on Garolite and ABS on PEI needs different first layer thicknesses for best adhesion, and both are more than 0.1mm.

BagelOrb, 2015-04-22:
I can't imagine the thickness of the first layer determines adhesion properties, but I'm no expert. Are you sure you are not just compensating the bed height by hand so that you are effectively changing the flow instead of the thickness of the first layer?

Renha, 2015-04-22:
I always follow this:
> If you tell your slicer to print a 0.20mm first layer, then it should be pretty darn close to 0.20mm.

It's №4 there http://forum.seemecnc.com/viewtopic.php?f=36&t=7361
So no, I'm not compensate flow ratio with thickness.

You could easily imagine how 1st layer thickness effects adhesion: when thickness is too low, you have high pressure there, and too strong adhesion (maybe not like here: http://forum.seemecnc.com/viewtopic.php?f=41&t=8065 but bad enough), when it's too high - too low pressure and no sticking. And different materials and different nozzle diameters needs different first layer heights

BagelOrb, 2015-04-22:
@Renha: Thanks for the post with some good insights!

I think you misunderstood me. When using the same gcode, of course the bed levelling influences the pressure and vica versa. I am talking about having the bed levelled at the correct height, corresponding to at which height you sliced the first layer of the model; that should not influence the pressure (much).
Given that for different first layer heights the pressure will be the same, would there still be differences in adhesion properties?

nophead, 2015-04-22:
It isn't the first layer height that affects adhesion directly it is the
width over height of the bead that determines how much it is squashed
against the bed.

If you increase the flow rate to more than the layer volume you get the
nozzle ploughing through the plastic, so the volume should always be
correct for the layer height.

Using the correct height and flow rate, but wider tracks will give better
adhesion.

BagelOrb, 2015-04-22:
So you are saying that perfect bed levelling and normal flow for the given layer height will result in the best adhesion?

I would say that a slightly increased flow can help in better adhesion, because of the increased pressure. It is only when you put the flow too high that the ticking etc. occurs.

nophead, 2015-04-22:
I always use the correct layer height and flow rate but when I print things with thinner layers through the same nozzle I use a higher W/T and things stick a lot more. Too much in fact sometimes and I end up breaking the glass to remove them.

I.e. you get more adhesion when the plastic is squashed against the glass. Just increasing the flow rate does that but if there is nowhere for the plastic to go you get a ridge around the bottom of the object and the nozzle has to plough through it until it sorts itself out after a few layers. Much better to have the correct flow rate to 100% fill the layer but use low wide tracks so the plastic has to be squashed flat from the cylinder it would naturally try to be.

BagelOrb, 2015-04-23:
@nophead: thanks for the advice. 

I will not introduce an initialFlowrate setting after all.

However, I am still thinking about the default settings. 
Is the default of 0.3mm initial layer thickness a good amount for PLA?
I would think it's a bit much.
Shouldn't we maybe decrease it to 0.2mm?

nophead, 2015-04-23:
It will stick better at 0.2mm if you keep the width the same, but of course it requires your bed to be more accurately levelled.

Renha, 2015-04-23:
0.2 is good for ABS too

closed 2015-04-28 by BagelOrb

daid, 2015-04-29:
The 0.3 is for uneven beds, variations in the bed levelness (wavy bed). But it's also often used for "overextrude" the first layer.

Which is another idea pretty much. You could have an "extra flow for first layer" setting, but then you also better give the first layer an extra offset to prevent the elephant foot.

BagelOrb, 2015-04-29:
I think we have agreed that the practice of adjusting the bed levelling for overextrusion is bad practice. That is also what I have read in a couple of blog posts on perfecting the initial layer. 

I therefore think that the initial layer offset is only enabling people to abuse their incorrect use of bed levelling.

Perhaps another way to increase the pressure at the first layer is to increase the line width a bit beyond the nozzle size - say 110%. That way we can pump up the pressure without overextruding. However, some new feeder mechanisms fail quickly if the pressure gets too high (Bas), so this might not be a good idea after all.