Ultimaker/CuraEngine #195: CuraEngine is missing protobuf symbols

oliv3r, 2015-05-07:
Log: /usr/bin/CuraEngine: symbol lookup error: /usr/bin/CuraEngine: undefined symbol: _ZN6google8protobuf2io17CodedOutputStream28WriteVarint32FallbackToArrayEjPh

CuraEngine will continue with the job at hand however.

supergis, 2015-08-05:
On Ubuntu15.04, Build CuraEngine got error:

Linking CXX executable CuraEngine
CMakeFiles/CuraEngine.dir/src/commandSocket.cpp.o：在函数‘google::protobuf::internal::GenericTypeHandler<cura::proto::SlicedObject>::New(google::protobuf::Arena*)’中：
commandSocket.cpp:(.text._ZN6google8protobuf8internal18GenericTypeHandlerIN4cura5proto12SlicedObjectEE3NewEPNS0_5ArenaE[_ZN6google8protobuf8internal18GenericTypeHandlerIN4cura5proto12SlicedObjectEE3NewEPNS0_5ArenaE]+0x73)：对‘google::protobuf::Arena::AllocateAligned(unsigned long)’未定义的引用

Would any suggestion for me? Thanks in advance.

BagelOrb, 2015-10-23:
This issue is so old that if it still were an issue, more activity would be expected here.

I therefore conclude that this is not an issue anymore.

closed 2015-10-23 by BagelOrb