Ultimaker/CuraEngine #196: Where is polygon.cpp in CuraEngine repository?

raceyi, 2015-05-08:
Hello Daid,

polygon.cpp is required as CMakeList.txt has it, right?

Would you please check CuraEngine repository?

I cloned it as below.
git clone https://github.com/Ultimaker/CuraEngine.git

and looked for polygon.cpp but it doesn’t exist.
find . -name polygon.cpp

How can I get polygon.cpp? or is it unnecessary to build CuraEngine?

Thanks in advance.

3dht, 2015-05-08:
I have same issue when using cmake:
I tried to change the line 55 to Polygon.h but not working

Here is error displayed: 

Make Error at CMakeLists.txt:55 (add_executable):
  Cannot find source file:

    src/utils/polygon.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

daid, 2015-05-08:
@BagelOrb Looks like you forgot to include that file when you added it to the CMake file.

daid, 2015-05-08:
Note, we merged a whole lot of development in the master branch, which changed everything from APIs to setting names, you might want to look at the legacy branch, which is the engine as it is distributed with the mainline Cura right now

raceyi, 2015-05-08:
Hello Daid,

Legacy branch doesn't have polygon.cpp. 
But I built and ran it successfully. 

It seems that legacy doesn't require protobuf.
According to your explanation, I had better use the master branch for further job.

Would you please check the master branch?
 
Thanks for your response in detail.

BagelOrb, 2015-05-11:
I'm sorry! Sometimes the most simple of things slip my mind.
Unfortunately it happened just before a long weekend, so the inconvenience was 'maximized'.
Sorry guys!

closed 2015-05-11 by BagelOrb