Ultimaker/CuraEngine #197: CuraEngine 14.06 and more(?) lowers bed before moving after extruding the primer.

Toffe87, 2015-05-11:
A newly generated gcode does this:

G1 Z0.20
G0 F9000 X115.97 Y95.10 Z0.15
;TYPE:WALL-INNER
G1 Z0.15

Lowering the Z before moving makes the nozzle drop right into the newly extruded plastic from the priming in the start. Can this be fixed?

Dim3nsioneer, 2015-05-11:
Are you sure you do not have an older version of the RetractWhileCombing plugin activated? I don't see such an issue with current Cura versions.

Toffe87, 2015-05-11:
No plugins, it happend after the last update.

Normal routine would be: 
1. Heat
2. Extrude 3mm plastic
3. Move to XYZ at the same time.

New routine is:
1. Heat
2. Extrude 3mm plastic
3. Move Z to 0.03
4. Move to XY

That last routine creates an drag on the 3mm plastic which is beeing dragged around the printbed.

Toffe87, 2015-05-12:
With BRIM it did not do that.

http://link.jepb.no/axjl - Without brim  (totally wrong and fcked) 
http://link.jepb.no/axD4 - With brim, (Correct way)

BagelOrb, 2015-05-13:
Verifying...
With brim: no problem
With zero brim perimeters: no problem
With skirt: no problem

I'm currently configured for an UM2, what settings do you use?

More importantly: I cannot find the problem in the gcode segment you posted! What line is wrong according to you?

BagelOrb, 2015-05-13:
I cannot reproduce the problem.
Can you please submit your command line arguments?
More specifically, what flavour did you use?

closed 2015-05-13 by BagelOrb

Toffe87, 2015-05-13:
Try disabling the lines(skirt) set it to 0?

It's at line 25 on that correct one and 26 on the wrong one. Looks like it's because of skirt is on 0?

BagelOrb, 2015-05-13:
Perhaps you are right, but I will not put any effort in verifying the problem for the old version of Cura, since it doesn't occur in the new Cura.

Toffe87, 2015-05-13:
I just updated a few weeks ago? (cura on Mac) Is that old version? The problem can still exist in the new if you haven't done any changes? The problem just don't magically disappears when you create a new version number! :-P

BagelOrb, 2015-05-13:
The new Cura isn't released yet.
A lot of work has been done since the old Cura.
As I said before: the problem doesn't still exist. 
Call it magic if you will. :0

daid, 2015-05-13:
a) We're not seeing the problem you are describing.
b) You fail to provide proper version numbers (14.06 is hardly new, as it's almost a year old)
c) You also don't provide a "I see this, and I would expect this" bit of info.
d) And you fail to provide your configuration settings.
e) We totally reworked the engine code for the next major release, which means your problem is most likely gone already.

Toffe87, 2015-05-13:
a) Probably because you havent removed skirt (0 in settings) as described above + no brim/raft. (See post #6)
b) 14.06 is not new, i misunderstood and thought cura was something different than cura engine, but its 15.02 i have 14.06 was someone else who also saw the bug also!)
c) I provided a "I see this and i would expect this" in post number #3 ;)
d) Because i did not find a typical configuration at first which i would expect doing this, until i remembered i had altered the configuration by removing skirt (0 on line count). My bad not doing a screengrab anyway and upload. 
e) I love that! Also I've talked to someone on facebook about this, so its understood and I've enabled 1 line of skirt anyway so its fixed until next rls! :)

For reference: IF you want to reproduce this in 15.02: Disable BRIM, RAFT and SKIRT then it magically adds on "LAYER 0" this:
G1 F2400 E-4.50000
G1 Z0.500

Without skirt image: (Bug highlighted): http://link.jepb.no/ay7Z
With skirt: (Bug no longer between the highlighted lines): http://link.jepb.no/ayZ2

So thanks a lot for a great engine and i cant wait to the new major release! :D