Ultimaker/CuraEngine #198: library mode

thawkins, 2015-05-12:
Would it be feasable to create a library target to create llbCuraEngine, essentialy a version of the codebase in a library without the main function. 

I have a bunch of utilities i want to write for generating gcode that could use an architecture identical to the gcodeExport and gcodePlanner stages, it would be cool not to have to reinvent or cut and paste that whole section of code. 

Alternativly convert the application into a main stub that links to a library. 

If i where to create an alternative CMakelists.txt to create a library, would be of interest and would it be worth pulling.

thawkins, 2015-05-12:
The following hacked CMakeLists.txt file builds a shared lib. 

1. I split the sources into app_SRCS and engine_SRCS
2. new shared library target libCuraEngine using only engine_SRCS
3. removed clipper library build, and just added it to engine_SRCS, its only a single .cpp file so there seems to be little point in building it as a library, it allows the engine sources to be compiled untouched as a shared library, and eliminates the need to link against it when building utility apps. 

Tests still pass on the CuraEngine Target, if this is of interest i will create a pull request. 
sts still pass on the CuraEngine Target, if this is of interest i will create a pull request. 
===============================================================================
project(CuraEngine)

cmake_minimum_required(VERSION 2.8.12)

find_package(Arcus REQUIRED)

if(NOT ${CMAKE_VERSION} VERSION_LESS 3.1)
    set(CMAKE_CXX_STANDARD 11)
else()
    set(CMAKE_CXX_FLAGS "-std=c++11")
endif()

# Add warnings
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall")

include_directories(${CMAKE_CURRENT_BINARY_DIR} libs)

#add_library(clipper STATIC libs/clipper/clipper.cpp)

set(engine_SRCS
    src/bridge.cpp
    src/comb.cpp
    src/commandSocket.cpp
    src/gcodeExport.cpp
    src/gcodePlanner.cpp
    src/infill.cpp
    src/inset.cpp
    src/layerPart.cpp
    src/mesh.cpp
    src/multiVolumes.cpp
    src/pathOrderOptimizer.cpp
    src/polygonOptimizer.cpp
    src/raft.cpp
    src/settingRegistry.cpp
    src/settings.cpp
    src/skin.cpp
    src/skirt.cpp
    src/slicer.cpp
    src/support.cpp
    src/timeEstimate.cpp
    src/Weaver.cpp
    src/Wireframe2gcode.cpp

    src/modelFile/modelFile.cpp

    src/utils/gettime.cpp
    src/utils/logoutput.cpp
    src/utils/polygonUtils.cpp
    src/utils/polygon.cpp
    libs/clipper/clipper.cpp
)

set(app_SRCS
     src/main.cpp
)

protobuf_generate_cpp(engine_PB_SRCS engine_PB_HEADERS Cura.proto)

add_executable(CuraEngine ${app_SRCS} ${engine_SRCS} ${engine_PB_SRCS})
target_link_libraries(CuraEngine Arcus)

add_library(libCuraEngine SHARED EXCLUDE_FROM_ALL  ${engine_SRCS} ${engine_PB_SRCS})

if (UNIX)
    target_link_libraries(CuraEngine pthread)
endif()

set(test_SRCS
    src/test.cpp
    src/utils/polygonUtils.cpp
    src/utils/gettime.cpp
    libs/clipper/clipper.cpp
)

add_executable(Test ${test_SRCS})
target_link_libraries(Test Arcus)

nallath, 2015-05-13:
Yeah, it should be possible. Not sure if its that interesting for us. Were are in the process of re-factoring a lot of stuff. In principle you could always use the application with the arcus socket, justlike the new GUI does.

BagelOrb, 2015-05-13:
You don't have to copy/paste our code, since you can just use the CmakeLists file you yourself have. It doesn't need to be in the repo.

However, as a shared library, the same library can be used in executing Cura and your tools.

Still I don't think it's smart to publish (part of) our code as a library, since we will probably quite frequently be breaking the interface, because there are still quite some refactoring steps to make.

closed 2015-05-13 by BagelOrb

daid, 2015-05-13:
(Also, a library means all your other code will get infected with AGPL. Which can be a good or a bad thing depending on your standpoint)

BagelOrb, 2015-05-15:
I say it's a good thing! :D