Ultimaker/CuraEngine #203: Extruder values messed up when built on armv7l

andrewhodel, 2015-05-18:
I built CuraEngine on 2 different architectures, here are the results of each.  Notice the E values.

**Linux 3.18.0-trunk-rpi2 #1 SMP PREEMPT Debian 3.18.5-1~exp1.co1 (2015-02-02) armv7l GNU/Linux**

```
./build/CuraEngine -o test.gcode ../10mmCalibrationSteps.stl

;Generated with Cura_SteamEngine DEV
;LAYER:0
M107
G1 F9000 X-0.313 Y-3.174 Z0.300 E264495377981531127500093123674537785986943011627008032420622878534344740506753690713281547055939670602955489149452884427009981847002875214639924230874178681599679896271274796406145054954071373746534621689559239516581262220836120258570337278267265493064256019553779541225955482270102996062784238565457920.00000
G1 X0.998 Y-3.200 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
;TYPE:SKIRT
G1 F900 X39.326 Y-3.200 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.069 Y-3.190 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.198 Y-3.184 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.242 Y-3.181 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.377 Y-3.168 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.444 Y-3.160 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.523 Y-3.147 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.631 Y-3.128 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
G1 X40.685 Y-3.116 E372759605167374253274767923800820318687286871498173892487448381845851951317432854574908620895844404130265558327485897256569517204897903039907201159682295978275143693857944591269564000650419053730762147950334039870853426830077573855182324935565186592105185645759543773849846270000717801129860408683790336.00000
```

**Linux debian 3.2.0-4-amd64 #1 SMP Debian 3.2.68-1+deb7u1 x86_64 GNU/Linux**

```
./build/CuraEngine -o test.gcode ../10mmCalibrationSteps.stl

;Generated with Cura_SteamEngine DEV
;LAYER:0
M107
G0 F9000 X-0.313 Y-3.174 Z0.300
G0 X0.998 Y-3.200
;TYPE:SKIRT
G1 F900 X39.326 Y-3.200 E4.59936
G1 X40.069 Y-3.190 E4.68853
G1 X40.198 Y-3.184 E4.70402
G1 X40.242 Y-3.181 E4.70932
G1 X40.377 Y-3.168 E4.72559
G1 X40.444 Y-3.160 E4.73369
G1 X40.523 Y-3.147 E4.74330
G1 X40.631 Y-3.128 E4.75646
```

protobuf, libArcus and CuraEngine were built using the same commands on both architectures.  It is strange that all of the other math is being done exactly that same, but the extruder math ends up incorrect on arm.

Please fix, I can build/test any updated changes or provide a system to ssh into for building on arm if needed.

This was the same when built with branch 15.06 or master on both machines.

daid, 2015-05-18:
I've been building for ARM with hardware float without issues, you might want to check if this isn't a compiler issue on your side.

oliv3r, 2015-05-18:
Are you cross-compiling or natively compiling on arm? For cross-compiling make sure to apt-get the gnueabihf variant, arm-linux-gnueabihf-gcc ( 4.9.2-10) 4.9.2 for example. Natively on arm it should be the proper compiler by default

andrewhodel, 2015-05-18:
Natively compiling on Arm, what I don't understand is why all of the other math seems to work fine.  Surely if it was a hard/soft float issue it would show up on the math for the paths.

daid, 2015-05-18:
The paths is 90% integer math, the extruder mostly floats.

andrewhodel, 2015-05-18:
When built with gcc/g++ 4.8 (not 4.9) it works on Debian Jessie for Arm.

```
apt-get install gcc-4.8 g++-4.8
rm /etc/alternatives/cc
rm /etc/alternative/c++
ln -s /usr/bin/gcc-4.8 /etc/alternatives/cc
ln -s /usr/bin/g++-4.8 /etc/alternative/c++
```

Still strange.

closed 2015-05-18 by andrewhodel

oliv3r, 2015-05-18:
I think your 4.9 version might be slightly out of date. I know there was a regression at some point and it was fixed. I'm not sure if it went into the regular jessie repo though, which could explain your 4.8 version working. I've also worked with the http://www.emdebian.org/ repositories as those tend to be slightly more up to date for the toolchains. Well worth to check it out.