Ultimaker/CuraEngine #204: what's the g-code for color?

columbus1231, 2015-05-19:
I am wondering how color is represented in g-code after conversion from an amf file. 
For example, just like a straight linear movement is represented in g-code as G1 Xnnn Ynnn Znnn Ennn Fnnn Snnn. How is the red green blue (rgb) color code read in g-code by a 3d printer in a section?

daid, 2015-05-19:
There is no color in gcode. (And there is no S in a G1)

columbus1231, 2015-05-20:
Thanks for your help. The S was a typo. LOL.

fieldOfView, 2015-07-23:
Apparently, there is an S in a G1:

closed 2015-08-27 by BagelOrb

Renha, 2016-02-17:
Hi @columbus1231 !
Color representation is done with M163, look here:
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/doc/Mixing%20extruder%20theory.txt
and more generally here:
http://reprap.org/wiki/Repetier_Color_Mixing

mvpolo2k3, 2016-09-18:
Renha and GitHub,

I have a machine that was a desktop CNC, work area 24"x24" that I am converting to a 3d printer. I have a 3 filament extruder head already, and 3 stepper drivers that are "cold" and drive the filament remotely. They are hooked up to axis A,B,C on my stepper drive. I manually go into my G-code and change "E" to A,B, or C to change color. I can get it to print in a singe color, since the "E" axis controls the feed for the filament. But my questions is similar to the OP, can I change the axis designation for each stepper drive for each filament. I dont plan on "mixing" colors, but I prefer to use black for infill and a solid color for outfill.

IE
Blue on extruder 1 is axis A
green on extruder 2 is axis B
red on extruder 3 is axis C

Is there a script or any other way to make it work where when it changes colors it changes "E" to A, B or C automatically?

BagelOrb, 2016-09-18:
What type of firmware do you have that used A B C rather than E?

daid, 2016-09-18:
MACH3 GCode flaver (used) to do this?

BagelOrb, 2016-09-19:
That's the one!
Is that the type of firmware you have?

CuraEngine dots have special casing for mach3, but there is currently not a
single machine which uses mach3 gcode flavour...

mvpolo2k3, 2016-09-19:
So I messed around in Repetier last night a bunch, and yes im using Mach3 for exporting and running my machine, not a standalong like Arduino controller. So basic 5 axis stepper driver. The main reason im running this is since its an old CNC table, it has NEMA42 steppers that run 5A/phase and no board ive found runs over 2A/Phase. Its got 1.5" ball screws on each axis, and had a 5HP spindle, so a pretty robust machine. I have since bought a few ACTUAL CNC machines (Kitamura Mycenters), and dont really need this machine to run CNC now. 

SO heres what I plan to do. Since I dont need to RGB mix colors, mainly just print one filament at a time, heres what I found. Using Slic3r to cut my files, using the "Teacup" or "mach3" Gcode flavor, I get each filament as a "tool" T0, T1, T2, etc. This works great, since I can just modify a script in Mach3 to deactivate all aux outputs, and active a single one per tool called. Then, using a 4 contact double throw relay, switch the stepper output from my E (or A as mach3 calls it) to run whichever extruder stepper I need. This would basically give me indefinite extruder head options, barring how many additional outputs I could call with Gcode.

mvpolo2k3, 2016-09-19:
My first, original thought was to try and use axis A B and C on mach3, to run each filament extruder, and be able to RGB mix colors, but it looks like that isnt possible with this software, since it would have to step 3 axis at different speeds for the RGB, and move X and Y and mach3 cant do that.

BagelOrb, 2016-09-22:
In Cura 2.3 next beta and final release you can select MACH3 as firmware when you select a custom printer.