Ultimaker/CuraEngine #205: Example config/profile file?

Renha, 2015-05-19:
Some day after updating CuraEngine and OctoPrint same time i have all broken: it seems old cura profile files does not affect anything now. It could be OctoPrint fault, but i don't believe it cause @foosel is so nice.
By the way, even if profiles code is okay, example profile file would be very useful (for me - to fix issue without downgrading, for others - pros are obvious). Thank you.

foosel, 2015-05-19:
> but i don't believe it cause @foosel is so nice.

Well thank you, but that still doesn't rule out an error on my part. 

It would probably help both the maintainers of CuraEngine and me if you could be a bit more precise in regards to which versions you are running now of both pieces of software (cura integration between OctoPrints development and master branch are completely different for example) and provide the profiles that don't work anymore.

Renha, 2015-05-19:
Oh I'm sorry... I've used current devel OctoPrint and master CuraEngine, but I am not sure it was last revision or as it mas on 13 may 2015 (sorry, I'm not very experienced git user and i don't know how to make link). I'll check it one more time with 100% latest revisions and would put profile files here if problem still exists.

daid, 2015-05-19:
You want the legacy branch of CuraEngine, instead of the master branch. Pretty sure @foosel did not update Octoprint for the new settings/profile/interfacing for the cleanup we pushed to the master branch a while back.

foosel, 2015-05-19:
@daid can you summarize what changed?

Renha, 2015-05-19:
Thank you for answer, I will jump to legacy for now. Please sorry for bothering both of you.

daid, 2015-05-19:
@foosel everything? :-)

Setting names are loaded from a json file (which is shared between the GUI and Engine)
Advanced socket communication is overhauled, and uses more "shared code" (libarcus, protobuf3) which makes it easier to keep the python and C++ code in sync. Also allows for backwards compatible changes from this point on.
No more "dictionary of doom" (as we called it), which converted the user visible settings to internal settings. So no more giving settings in microns to the engine, the engine now works in mm, degrees and just sane units overall. And thus also accepts floats now.

If you are using the commandline interface (which still works fine) you just need to use the new setting names, supply the json file (which you can also use yourself, as it has names and tooltips for all settings). The rest will still work just fine.

foosel, 2015-05-19:
That kind of summary is enough, thanks ;) I'm indeed using the command-line
interface. A quick look into the sliceEngine module in Cura didn't look so
different concerning the settings names, but I'll take a closer look when
I'm back on my feet.

BagelOrb, 2015-10-23:
run `CuraEngine --help` to see how to use the new backend as standalone.

I think it would be better to connect to the backend via the command socket than passing the settings as command line arguments. Take a look at Ultimaker/libArcus!

closed 2015-10-23 by BagelOrb