Ultimaker/CuraEngine #206: use assimp to handle file format import/export.

thawkins, 2015-05-21:
Assimp is a multilke format importer for mesh and material data, that reads a very wide number of file formats, and provides both import and output capability for most. It all provides a number of mesh quality and repair functions that can be used to normalise or fix up meshes. 

It would make a good replacement for the existimg input/output system on CuraEngine. 

http://assimp.sourceforge.net/index.html

.stl and .obj format is fully supported.

BagelOrb, 2015-05-26:
Might be a good idea. We should take a look into this.

Ghostkeeper, 2015-05-27:
This is an issue for the front-end, Cura, rather than CuraEngine.
That said, I'm not sure if depending on Boost is a good idea, but it might be a good idea to use this or something similar.

nallath, 2015-05-27:
I agree with Ghostkeeper, depending on boost should be avoided if possible.

daid, 2015-05-27:
Boost? That's really something we want to avoid.

Also see #208 for the reason why this does not belong at the engine.

closed 2015-05-27 by daid

awhiemstra, 2015-05-28:
In addition, we have been considering implementing a file reader/writer plugin for Cura that uses AssImp since there are also Python bindings available.