Ultimaker/CuraEngine #207: CuraEngine does not prepend machine_settings.machine_start_gcode.default to the outputted gcode file

andrewhodel, 2015-05-21:
Simple enough to fix in the UI, but this should really be done by CuraEngine.

While on the subject:

categories.material.settings.material_print_temperature.default
categories.material.settings.material_bed_temperature.default

Don't get prepended to the gcode either.  I guess that's left up to the UI?

daid, 2015-05-21:
https://github.com/Ultimaker/CuraEngine/blob/master/src/FffGcodeWriter.cpp#L150

closed 2015-05-21 by daid

andrewhodel, 2015-05-22:
This was for branch 15.06 which does not use command line options, what is the stable branch?

Is master to be used as stable?

daid, 2015-05-22:
No, legacy is most likely the most stable branch. 15.06 is the release branch in which we are bugfixing. master is the development branch.

andrewhodel, 2015-05-22:
On master it still doesn't write 'M109 S{temp}' to warm up the extruder if you set machine_gcode_flavor.default to something other than UltiGCode like RepRap.

https://github.com/Ultimaker/CuraEngine/blob/master/src/FffGcodeWriter.cpp#L144

https://github.com/Ultimaker/CuraEngine/blob/master/src/gcodeExport.cpp#L394

It does however correctly write the machine_start_gcode when machine_gcode_flavor.default is set to something other than UltiGCode.

Is Cura tracking the temperature in the slicer with Ultimaker/UltiGCode?  That seems out of place...

BagelOrb, 2015-05-26:
Why is the machine_start_gcode not prepended by the engine for UltiGCode?
Is it still done by the front-end?

daid, 2015-05-26:
Because UltiGCode is strange:
https://github.com/Ultimaker/CuraEngine/blob/master/src/FffGcodeWriter.cpp#L57

BagelOrb, 2015-05-26:
When not using the command socket, the starting gcode isn't prepended. I will fix this. 
https://github.com/Ultimaker/CuraEngine/commit/9d31a525c9c42af77b617637767ab41fb9d20fe5

daid, 2015-05-26:
Pretty sure that commit will get double prepended code now in the socket communication.

Also, using -1 for material will not work for the firmware, as it won't heat the hotend.

BagelOrb, 2015-05-26:
Just did a print with it, no heating problem.

Sorry for the rash commit. 

See https://github.com/Ultimaker/CuraEngine/commit/e8d117d62fef50a61dbf67ed177a2e51f8712574