Ultimaker/CuraEngine #208: 3MF Support

PatternSpacePrime, 2015-05-22:
Are there any plans in the works to support the new 3MF file formats?

Dim3nsioneer, 2015-05-22:
+1 for 3MF support (it happens quite rarely that big players agree early on a file format)

daid, 2015-05-24:
Not directly in the engine. The engine is passed 3D data from the frontend in a uniform way. The STL loader is just to assist in debugging.

So Ultimaker/Cura will most likely get 3MF support at some point (even tough Microsoft initially said it was only an internal exchange format when I got air of it a year ago)

PatternSpacePrime, 2015-05-25:
Daid, I tried looking for your email to contact you directly, but was unable to find it so I'm going to post this here: If I wanted to contribute to the project by adding 3mf support to Cura, should i do it in the Engine or the front end project? And if front end, what's the difference between the Ultimaker/Cura repo and the Daid/Cura repo? And if I use the Daid repo, what branch should i use? PinkUnicorn? SteamEngine? Master? 

Thanks.

daid, 2015-05-25:
d.braam@ultimaker.com is my email. But I rather keep most discussions public :-)

Daid/Cura is the legacy (currently released) Cura, and works with the Ultimaker/CuraEngine#Legacy branch. No new development is done on this, only bugfixing.

Ultimaker/Cura is the new development. Currently in beta testing. It uses Ultimaker/CuraEngine#master branch, and new developments should go here.

PatternSpacePrime, 2015-05-25:
Thank you, Daid. I will close this issue and start looking into contributing to the Ultimaker/Cura Project.

closed 2015-05-25 by PatternSpacePrime

BagelOrb, 2015-05-26:
This issue might be of interest: https://github.com/Ultimaker/CuraEngine/issues/206