Ultimaker/CuraEngine #214: Stutter in spiralized vase

CCS86, 2015-06-03:
When using the spiralize outer feature, and unchecking solid top, on a solid vase model in cura, the output code has bad moves. The layer view looks perfect, but in roughly the same spot on each lap (but not every single lap) Cura makes a little stutter move with the print head. It basically moves backwards 1-2mm then continues forwards. It is almost hard to see with your eye unless you are looking for it, but the print shows a telltale scar. This occurs at the "layer change" area of the code.

[image]

[image]

[image]

Gcode:
https://dl.dropboxusercontent.com/u/60958586/Doggy%20Vase.gcode

Forum thread:
https://ultimaker.com/en/cura/view/16189-cura-1504-spiralized-vase-stutter-scar?page=1#reply-108615

BagelOrb, 2015-06-03:
Did you encounter this problem on the old cura or in the new CuraEngine?

CCS86, 2015-06-03:
Good point, this is on Cura 15.04, so I'm not sure it applies to the new engine or not. A moderator on the UM forum liked me here to submit it.

nallath, 2015-06-04:
15.04 is the old engine.

BagelOrb, 2015-06-04:
I think the Spiralize option is bugged in the new Cura as well.
Can you maybe provide a link to the stl? I am in need of a typical spiralize model.

CCS86, 2015-06-04:
https://dl.dropboxusercontent.com/u/60958586/Doggy%20Vase.stl

CCS86, 2015-06-04:
Here is the same model sliced in the new engine. The bug still exists, but now instead of a G1 move backwards ~once per lap, it is a rapid move forwards:

[image]

BagelOrb, 2015-06-04:
This is a different bug.
The move forward is most probably due to the fact that each layer a new starting point is selected, which is closest to the old one. Perhaps there is no other vertex closer. It would then be an enhancement to find the closest point on a line segment, instead of the closest vertex of a polygon in the next layer. However, such optimization might be a considerable hit on performance, since we need to interpolate on each line segment.

The other possibility is that the wrong point is selected while there is a vertex that is closer. This would be a bug.

I will need to dig into this in order to solve it.

Perhaps you can easily see whether the points in the next layer contain a point which is closer to the end point of the previous layer?
If I would know that I could easily disambiguate between the two cases.

CCS86, 2015-06-05:
In a spiralized print, there is no need for a true "layer change". I know that Cura wants to segment the print that way, which is fine. In that case, why wouldn't the "layer" start at the same point it ends? It shouldn't need to find different vertices for layer start and end points.

daid, 2015-06-05:
Because the whole spiralize implenentation is a huge hack.

BagelOrb, 2015-06-05:
Yes, I am not going to rework the whole slicer engine for a correct way to print spiralized prints.

CCS86, 2015-06-05:
Why would you have to rewrite the entire engine? 

Spiral vases seem like a pretty legitimate type of print. You guys just don't care?

daid, 2015-06-05:
> You guys just don't care?
Yes. (Honesty is hard)

CCS86, 2015-06-05:
That didn't seem very hard for you. If you want to let a relatively simple programming task defeat you, so be it. Others have solved it.

daid, 2015-06-05:
We hate relatively simple tasks. But you are free to send us a bunch of money so we actually can get paid to fix this for you.

BagelOrb, 2015-06-05:
Oh that diad.
Spiralize is not a high priority atm. There are more pressing issues.
Moreover, spiralize is working quite fine now.

Note that the issue is still open: 
The layers should have smaller connections; there are vertices closer to the last point on the previous layer. This might just be an issue concerning non-spiralize mode as well.

CCS86, 2015-06-05:
Hahaha, Daid, I don't remember you being this bitter! I have sent you guys a bunch of money for my printer, supplies, HBK, etc. I would have even bought Cura as well, but you sold me junky-ass netFabb instead!

Thanks Bagel, glad it's still on the radar. How do you figure that it is working "quite fine" though?

nallath, 2015-06-05:
Cura is not owned by ultimaker, just heavily supported by it.

daid, 2015-06-05:
And I have a tendency to get grumpy and mean when people act like I haven't done shit for them, while they have been using my software, for free, for years. And then suddenly complain that I won't make major changes for them for a minor feature.

CCS86, 2015-06-05:
Wow, you sure "read" quite a lot in between the very few lines I wrote. Definitely not sure which part implied to you that I think so little of you, or your contributions.

I have used, on and off, Cura for years, and have tried to be an active member in regards to diagnosis and feature suggestions for that time. Could you really get a piece of software like this fully sorted without any help from users? It seems strange to work hard on a piece of software intended for free distribution, then treat users as if they are freeloading assholes for using it and wanting it to be the best it can be.

Either charge for it, open an official donation system, or distribute it for free and be happy that people appreciate your work.

nallath, 2015-06-05:
" That didn't seem very hard for you. If you want to let a relatively simple programming task defeat you, so be it. Others have solved it." <- Im going to interupt your train of thought here.

Before that statement, we, as the developers, told you that this was going to be hard. Yet you decided to add an insult. Is it really that strange that people respond like that?

BagelOrb, 2015-06-05:
This is getting quite out of hand.
Snap out of it! :P
This is not the time nor the place for argue.

Anyway, I am still interested in the bug which is brought to light here: the first vertex on the next layer is not the one closest to the endpoint of the previous layer.

I will have to check whether it also occurs in non-spiralzie mode.

BagelOrb, 2015-06-05:
PS: the reason I deleted the discussion is that this is an issue tracking system and I don't want this bug to get lost in the discussion.

gr5, 2015-06-06:
This bug occurrs in non-spiralize also.  I'm 90% sure.  So it's not related to spiralize so much that it's more obvious.

gr5, 2015-06-06:
I don't think this is a bug.  The vase is broken into triangles and it's possible that none of the lines in these triangles are vertical such that when you pick the "closest point" on the next layer up you probably have to get offset by roughty the amount of layer height (if the triangle is tilted 45 degrees along its edge).  Even if you could tell your CAD to have as many vertical edges as possible in your STL file, when you slice these to lines in a layer, half the points will be on a vertical edge and half on a tilted edge.  Some CAD programs are probably worse about this than others.  The simplest fix I can think of is to predetermine all the layer change points before you slice - try to pick a path up the side of the vase that is as vertical as possible - this is kind of a traveling salesman problem so not easy but not unreasonable either as you can just find "a" good path, not the "best" path.  Still quite a bit of code to write!

Ghostkeeper, 2015-06-06:
I tried reproducing it in Cura 15.04. For this I made a sphere in OpenSCAD with the command:

    sphere(r=10,$fa=0.5,$fs=0.5);

[Here is the resulting STL file export.](http://dulek.net/Sphere.stl)

When using spiralise, I was able to reproduce the bug CCS86 shown would occur in the *new* engine, with the little G1 move once per lap causing gaps. I couldn't find any of the moves outside of the model as in the original bug report. When not in spiralise mode I didn't get such bugs but I found something that may be related and made a screenshot of it:

[image]

See the little upwards move on the right there. I also remember seeing this in @BagelOrb's tests of the improved combing. This one occurred in Sphere.stl, layer 224 when slicing with layer height 0.06, shell thickness 0.4, bottom/top thickness 0.6 and no support (the rest of the settings on default).

They are also travel moves, but they seem to be far too long to travel to the next layer.

CCS86, 2015-06-07:
Other users are finding evidence that this isn't a bug specific to the spiralized option, but with the layer start point logic, it is just much more apparent with that type of print. It would seem that if this is true, improving the layer start point logic could help all types of prints.

If I understand correctly, we have ended a layer at a vertex of a line segment, where a facet intersected with the current XY plane. Now we need to connect that point, with the next slice.

Could we then find the closest vertex on the next layer, which is generally in the same direction as the last print move? You could find the closest two vertices, one should be "ahead" of the nozzle, and the other "behind". Then, look at the average delta X and delta Y over the last 'n' number of print moves, and see which of the two vertices has deltas in the same direction(s). The first move of the next "layer" is then a linear print move from the ending vertex to the closest forward vertex.

BagelOrb, 2015-06-08:
@Ghostkeeper, what you see there is a retraction.

@CCS86: I should indeed think it would be as simple as that - or even more simple: we should only look at the very last move direction.

CCS86, 2015-06-08:
@BagelOrb : My thought behind looking at an average was that you may have been moving primarily in say X+, and meandering in Y. Maybe, the Y moves were mostly positive, but the last couple crossed over into Y-. Looking at the average could differentiate between these.

It's hard to say whether this would actually be a benefit or not, as it's hard to picture every possible situation that comes up with the wide variety of models people slice (something I'm sure you guys struggle with regularly). At least writing the code in as such would allow for testing different sample lengths to see any practical benefits/drawbacks, with the ability to set n=1 to look at just the last move.

CCS86, 2015-07-04:
Hi guys, glad to see the new version made its official release!

I tested the spiralized function again, and found it to be similar to my report above. I don't see any real issue with the identification of the next set of coordinates when "changing layers", which is good because that is the most involved process.

Basically as I see it, the only issue is the replacement of a print move, with a dog legged rapid. Cura posts (in this case) a ~2mm rapid move, followed by a much smaller rapid, then resumes printing. If we could just get it to make a printing move to that final coordinate set instead, I think there would be no issue:

Code with bug:

G1 X87.188 Y140.246 Z2.699 E671.56797
;LAYER:24
G0 F12000 X85.280 Y139.347
G0 X85.246 Y139.441
;TYPE:WALL-OUTER
G1 F1500 X84.361 Y139.015 E671.57413

Fixed:

G1 X87.188 Y140.246 Z2.699 E671.56797
;LAYER:24
G1 X85.246 Y139.441 EXXX.XXXXX
;TYPE:WALL-OUTER
G1 F1500 X84.361 Y139.015 E671.57413

Unfortunately, since the extrusions are being posted in absolute mode here, retroactively fixing this code would be fairly clunky.

Here are a couple pictures to illustrate:

[image]

[image]

CCS86, 2015-07-04:
[image]

CCS86, 2015-11-22:
the latest 11-17 build still has this same behavior.

BagelOrb, 2015-11-23:
It's not exactly the same in 11-17. As a quick fix, when spiralize is enabled, the travel speed is set to the print speed, so that the hysteresis effects of the bowden tube will cause there to be less of a stutter.

Also you can set the Z seam alignment to [ Back ], so that the distance between the starting points of different layers is very small.

CCS86, 2016-05-08:
I did some testing in the latest release. 

With Z seam set to shortest: 
[UM34_DoggyVase2-shortest.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/254046/UM34_DoggyVase2-shortest.gcode.txt)

There is a small rapid backwards move after each lap. It looks like just deleting that move could help, but would still probably leave a small artifact, as there appears to be a slight XY shift:

[image]

[image]

[image]

CCS86, 2016-05-08:
Other layers have a slight forward jump:

[image]

Setting Z seam to back looks very similar.

CCS86, 2016-05-08:
There is some very bizarre stuff going on in the bottom layer region though. Maybe there is a better setting than I am using:

[image]

CCS86, 2016-05-08:
I tried switching to concentric for the bottom layers, I have all gap filling and wall overlap comp disabled.

In this gcode, it seems to get through the bottom layers, start spiralizing, then jump to printing non-spiral, bad moves at layer 155:

[UM34_Doggy Vase-conc.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/254052/UM34_Doggy.Vase-conc.gcode.txt)

[image]

BagelOrb, 2016-05-08:
The visualization bug of RepetierHost is llikely due to a bug in the gcode. I've seen this sort of thing happening with negative E-differences from one line to the next. *Please* attach the model and export your profile - preferably in a new github issue. This is quite a serious bug and might occur not only in spiralize vase mode.

BagelOrb, 2016-05-08:
As for the travel moves: a small move in the XY direction is hard to avoid, but we could try replacing it with an extrusion move.

If you send me your profile I could slice the doggy vase with the option implemented and let you have a sneak peek at how it works.

CCS86, 2016-05-08:
Sorry, if that wasn't clear. That huge artifact in the background of the last picture is just the header gcode extruding some stock on the way in.

The issue I'm seeing is a transition from bottom layers, to spiral, back to multi-perimeter, non-spiral layers. 

Happy to create a new issue if you'd like though.

BagelOrb, 2016-05-08:
Ah alright. Yes, strictly speaking that is a separate issue. This thread is already getting quite long ;)

Shouldn't be that hard to fix, though.

CCS86, 2016-05-08:
Okay, I split off that other bug into a new issue. 

There is still a small "stutter" bug in the spiral regions.

CCS86, 2016-05-24:
I just wanted to update that with 2.1.1, the behavior is still there.

[image]

CCS86, 2016-05-24:
Choosing random seam, is pretty ugly:

[image]

CCS86, 2016-05-24:
It seems to me that this move just represents the accumulated error of the XY positions at one layer versus the next. This means it is governed by overhang angle, and layer height. Thicker layers and larger overhangs exaggerate it.

.1mm layer height:

[image]

.2mm layer height:

[image]

It's hard for me to give a useful suggestion without knowing the algorithm. 

Maybe the front end could get the engine to slice at ~1/10 the commanded layer height, and use that to distribute the error more evenly?

Or compare the closest XY coordinate between subsequent layers, determine an XY distance, then interpolate using current Z height and apply preemptively?

CCS86, 2016-05-24:
.02 mm layer, in a more vertical section of model: 

[image]

CCS86, 2016-05-24:
Besides the XY shift, I feel like it should never do a backwards move like this. It should always move forward.

BagelOrb, 2016-07-04:
The spiralize vase is a dirty hack. It just gradually increases z over each
polygon.

There is no (easy) way to avoid xy moves.

I don't think the backwards moves are high priority enough to put it on our
to-do list right now..

BagelOrb, 2016-07-05:
@gr5 please make a separate issue for that. Could you include some screenshots of what happens and say what you think should happen? 

Note that Cura will try to get the closest point, but also the point which is mostly in the same direction as it was travelling before, so that the first point is most optimal w.r.t. accelerations and jerk.

BagelOrb, 2017-03-08:
Thanks to Mark Burton, we don't have a stutter in the spiralize mode any more!
:clinking_glasses:

closed 2017-03-08 by BagelOrb

mattgit, 2019-01-05:
> I just wanted to update that with 2.1.1, the behavior is still there.
> 
> [image]

still there on cura 3.6.0