Ultimaker/CuraEngine #217: CuraEngine "master" crash, CuraEngine 15.06 works fine

MiguelAngelLV, 2015-06-19:
I install Cura from git using the Ubuntu Script, but CuraEngine crash when try slicer, the app restart curaengine but crash again.

If I change to 15.06 branch of CuraEngine (only), the app works fine.

BagelOrb, 2015-06-19:
Can you provide the STL file you were trying to slice? 
Did you use the same branch in all repositories? Eg all on master or all on 15,06

MiguelAngelLV, 2015-06-19:
I try with varius files. This for example:
http://www.thingiverse.com/download:598358

Yes, all repositories was "master", after change only CuraEngine to 15.06
because of with master not works.

MiguelAngelLV, 2015-06-19:
Now 15.06 crash too.

The dmesg show

```
[18059.059511] traps: CuraEngine[13389] trap divide error ip:4f6473 sp:7fff5256bd10 error:0 in CuraEngine[400000+1ab000]
```

BagelOrb, 2015-06-22:
What if you call CuraEngine directly from the command line? (see https://github.com/Ultimaker/CuraEngine for instructions)

MiguelAngelLV, 2015-06-22:
```bash
link@Hyrule:~/Descargas$ CuraEngine newstrooper.stl 
Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
ERROR: Failed to load json file: fdmprinter.json
Unregistered setting wireframe_enabled
Unregistered setting machine_gcode_flavor
Unregistered setting retraction_count_max
Unregistered setting retraction_extrusion_window
Unregistered setting material_switch_extruder_prime_speed
Unregistered setting material_switch_extruder_retraction_speed
Unregistered setting machine_switch_extruder_retraction_amount
Unregistered setting layer_height_0
Unregistered setting layer_height
Excepción de coma flotante
```

It looks as _float exception_. Float overflow?

I try with fmdprinter.json default file (this repo)

```bash
link@Hyrule:~/Descargas$ CuraEngine -c /home/link/dev/CuraEngine/fdmprinter.json -o model.gcode newstrooper.stl 
Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Unknown option: c
ERROR: Failed to load json file: fdmprinter.json
Failed to load model: /home/link/dev/CuraEngine/fdmprinter.json
Unregistered setting machine_end_gcode
Unregistered setting speed_travel
maxObjectHeight : 0
```

It show _unknow option_

BagelOrb, 2015-06-23:
No, the problem is the earlier error "Failed to load json file: fdmprinter.json".
You should provide the json file as a command line argument.

As I said, see https://github.com/Ultimaker/CuraEngine for instructions.

closed 2015-06-23 by BagelOrb

MiguelAngelLV, 2015-06-24:
Ok. I change the config. CuraEngine by commands works, but crash from Cura GUI.

Ghostkeeper, 2015-06-25:
Perhaps this is a localisation error resulting from reading floating point values in languages with an uncommon radix?

MiguelAngelLV, 2015-06-25:
I am spanish. In spanish the decimal use " , " (comma), but my in cure I
use . (dot). I don't know if my system change any number...

cwxda, 2015-07-09:
Same problem, using C locale.
OS: Arch linux X86_64 multilib
Even the most simple model of a cube doesn't load and cura keeps being stuck on Calculating

BagelOrb, 2015-07-19:
What's the status of this bug? Is it still present on master?
I should introduce some logging in CuraEngine shich shows each setting used, so that it is more easily reproducible and/or tracable.