Ultimaker/CuraEngine #220: Could CuraEngine rotate and translate a model based on matrix input before slicing?

magillus, 2015-07-12:
It would be useful in use with octoprint when moving STL and slicing with Cura engine to move and rotate an object before slice.

BagelOrb, 2015-07-12:
As I recall it, it is possible to supply a 3x3 transformation matrix. Perhaps this functionality got lost along the way. If so I will fix this tomorrow.

Note that your front end (octoprint) can just as easily apply the matrix before sending the model data. I propose you send the data via the command socket, and not read the STL file all over again.

BagelOrb, 2015-07-12:
Come to think of it a 4x4 matrix would be more appropriate. However, translation is performed separately through settings named something like mesh_position_x.

nallath, 2015-07-13:
Even in the new version? The Cura GUI used to have a translation and a 3x3 matrix, the new Cura uses 4x4 matrices.

BagelOrb, 2015-07-13:
The only use case for this feature is for direct use via the command line; it is not really in our interest to implement this feature.

But CuraEngine is open source! 
If you want this feature, you should implement a 4x4 matrix in src/utils/floatpoint.h which is pretty easy given the already implemented 3x3 matrix. Then it should be applied at src/commandSocket.h/122 and src/fffProcessor.h/86.
If you've got this working, submit a pull request and it will be included in the next release! 
:)

closed 2015-07-13 by BagelOrb

BagelOrb, 2016-10-04:
This issue has recently been fixed:
CuraEngine [...] -s mesh_rotation_matrix="[[1,0,0],[0,1,0],[0,0,1]]" -l /path/to/mesh.stl [...]

elnazasd, 2017-05-30:
Hi Guys,
I use command line for executing Cura 2.5 in Matlab. I want to change layer height, part orientation and filling angle and providing different G-code.
I start to generate different json file for different layer height but I dont know how i must change the orientation? would you please tell me how I can do it?
commandline=[str ' slice' ' -j ' definition ' -s mesh_rotation_matrix="[[1,0,0],[0,1,0],[0,0,1]]" ' ' -l ' stl ' -o ' stroutput '.gcode'];
str is CuraEngine.exe file and deffinition is innovo_inventor' num2str(j) '.json which is generated for different layer height.
Thanks in advance
Elnaz

BagelOrb, 2017-05-30:
If I were you I would give the layer height as a command line setting as well, so `-s layer_height=0.1`.
Putting that setting in a json file has no added value.
Note that the engine doesn't calculate the value function defined for each setting, but just uses the `default_value`.

If you want to change the orientation you should change the `mesh_rotation_matrix`.
If you want to know more about how matrices work you should read up on something like https://en.wikipedia.org/wiki/Rotation_matrix#Basic_rotations

elnazasd, 2017-05-30:
Thanks for your response, for my case, the layer height is not fixed. I define a For loop to change each time the default value of json. I dont know that I can change the layer height in command line or not? Also, would you please tell me how I can rotate for 45° or changing the position of part from Flat to upright or on edge position?
Thank you

BagelOrb, 2017-05-30:
If you are changing the value each time it's better to just use one and the
same json file and just alter the setting by giving it a different value in
the command line call.

The 45 degree depends on what edge you want to rotate on. Please study the
wiki article for a bit. You probably need to fill in .707 at some places in
the matrix (half the square root of 2).

Ghostkeeper, 2017-05-31:
To get the rotation matrix to rotate around an axis, you have to apply Rodrigues' formula: http://electroncastle.com/wp/?p=39

Ghostkeeper, 2017-05-31:
So in this case it'd be `-s support_enable=True`.

simar1998, 2020-03-12:
How would I move the stl mesh to printer coordinates from STL file ? the -s mesh_position_y=150 does not seems to have an effect

BagelOrb, 2020-03-12:
It should have effect...

Perhaps you have `center_object` still set to true?

simar1998, 2020-03-15:
> It should have effect...
> 
> Perhaps you have `center_object` still set to true?

I have set that to false but It still does not move the mesh. Here is my command 

CuraEngine slice -v -p -j /home/pi/definitions/creality_cr10s.def.json -s layer_height_0=0.2 -s layer_height=0.2 -s line_width=0.4 -s infill_sparse_density=4 -s infill_pattern=gyroid -s wall_thickness=1.2 -s top_bottom_thickness=0.8 -s ironing_enabled=false -s retraction_speed=30.0 -s material_print_temperature=200.0 -s retraction_amount=3.0 -s material_initial_print_temperature=200.0 -s material_bed_temperature=60.0 -s material_flow=100.0 -s cool_fan_speed_0=100.0 -s cool_fan_enabled=true -s adhesion_type=brim -s speed_print=60.0 -s speed_travel=120.0 -s speed_support=40.0 -s speed_wall=60.0 -s speed_infill=80.0 -s speed_topbottom=40.0 -s support_type=everywhere -s support_enable=true -s support_pattern=grid -s support_angle=40 -s support_infill_sparse_thickness=50.0   -s mesh_position_x=100 -s mesh_position_y=100  -s center_object=false  -o /home/pi/test1230.gcode -l /home/pi/gear.stl  --next

It seems to have not effect on the gcode generation and I am using the latest version of the CuraEngine.

BagelOrb, 2020-03-16:
What if you specify those object settings **after** you load in the stl?
i.e. `-l [object] -s [object_settings]`

simar1998, 2020-03-16:
> What if you specify those object settings **after** you load in the stl?
> i.e. `-l [object] -s [object_settings]`

Unfortunately that has no effect. Does Cura move the process the mesh in order to apply the mesh transition ?

simar1998, 2020-03-16:
Ok now it works. Thank you. I had to remove --next from the command line and it worked.