Ultimaker/CuraEngine #223: Inconsistent fills in Cura 15.06.2

Phalconheart, 2015-07-21:
There seems to be inconsistent results depending on what value you enter for the infill amount. I had proper results at 30% and 20%, but at 25% the infill lines were randomly spaced and nowhere close to 25%.
[image]
[image]
[image]

This is not just a display bug as I actually printed a model with these odd random infills and it actually printed with the infills as displayed in Cura.

Any ideas as to why?

BagelOrb, 2015-07-22:
@Ghostkeeper that issue is not related.

BagelOrb, 2015-08-27:
I'm sorry I lost track of this issue. 

I've done a lot about the infill in the mean time.

However, I'd like to be sure I solved your bug specifically. Can you send me the STL?

closed 2015-09-08 by BagelOrb