Ultimaker/CuraEngine #224: 'wireframe_top_delay' ignored unless 'wireframe_strategy' is 'knot'

fieldOfView, 2015-07-22:
The "Wire Printing Top Delay" setting is supposed to insert delays after vertical wires, but setting it to a non-zero value doesn't insert G4 delays in the gcode.

Looking at https://github.com/Ultimaker/CuraEngine/blob/master/src/Wireframe2gcode.cpp, it seems that top_delay is only added when the Knot strategy is chosen. I don't know if this is intended, but it does seem to me that top_delay can also be useful with the other strategies.

BagelOrb, 2015-07-23:
This is intended. When your strategy is to compensate for the filament falling down, adding a delay only causes more sagginess which cannot be compensated for.

avenosa, 2016-03-16:
Cura 1.99.0-dev.20151117 Mac
Strategy choose: knot
Top delay not working.
No matter what delay is put on top or bottom delay, it always delays only at bottom.

BagelOrb, 2016-03-17:
Wireframe top delay was initially only meant for the knot strategy.

Wireframe printing currently has low priority, so don't expect us to
implement a top delay for other strategies any time soon.

Perhaps I will disable the setting when a different strategy is chosen, so
that it's less confusing to people..

fieldOfView, 2016-03-17:
> Perhaps I will disable the setting when a different strategy is chosen, so that it's less confusing to people..

Yes, please. 

I still think it is a shame Wireframe is not one of the top hitting features of the new release, because it could be one of the "gems" that new Cura has that legacy Cura doesn't. Its fun, its new, don't let it go to waste.

nickthetait, 2016-03-17:
I agree, wireframe is indeed a "next generation" level feature.

Ghostkeeper, 2016-03-18:
Maybe move it into an "experimental" category then? The string freeze is lifted for a few days now.

nickthetait, 2016-03-18:
Gcode previewer can't render wireframe yet right? In that case an "experimental" designation makes sense to me.

Billiam, 2019-08-06:
In Cura 4.2.1, `wireframe_top_delay` also seems to be ignored when using the `knot` strategy as well. No difference in gcode (other than the settings comments), when using knot and different top delay values.