Ultimaker/CuraEngine #226: CuraEngine standalone on EC2 instance

maxdeb, 2015-07-29:
Hi,
I am trying to use CuraEngine on a EC2 instance with Amazon Linux (CentOS) as OS.
I am able to run the build on the server it has been build on (with libarcus installed), but on other machines i get following error:
./CuraEngine: error while loading shared libraries: libArcus.so.0: cannot open shared object file: No such file or directory

Is it possible to package every dependency or is there a way to install CuraEngine from repo? Amazon Linux only has its own repos and repel it can download from..

maxdeb, 2015-07-29:
when I add the path to libArcus and libprotobuf to the LD_LIBRARY_PATH variable, I am able to start CuraEngine. However I would love not to do that, if that's possible?

nallath, 2015-07-30:
I'm kind of curious, why do you want to put it on an EC2 instance in the first place?

maxdeb, 2015-07-30:
We're experimenting with a cloud based slicer. As per terms of your license, we will off course open source all code.

BagelOrb, 2015-10-08:
Try to compile it statically.

closed 2015-10-08 by BagelOrb