Ultimaker/CuraEngine #228: Request: Retreat before retract

CCS86, 2015-08-08:
I am noticing cases when the extruder is positioned on an external perimeter and performs a retract move (when an island has only perimeters, no infill, for example). This obviously leaves a nice blemish on the part surface.

Would it be possible to ensure that the extruder makes a small rapid move away from the perimeter before initiating the retracted rapid move?

Thanks!

BagelOrb, 2015-08-10:
It used to be a feature that after printing an external perimeter the head moved toward inside the part by a mm or so. 

However, people who printed the outer perimeter only and people who used the spiralize outer perimeter option complained about this move. I think we explicitly disabled the feature for those cases.

Now before trying to reproduce the problem I am boggled about this request.
Don't you think such a feature will more likely rip apart the perimeter?
Wouldn't it also leave more oozed material and thus more scars?

CCS86, 2015-08-10:
Why would you be "boggled" by the request if it used to be a feature? I have not proven that it can work, but I believe there is a way to successfully make it work. 

For the issue of it not always being applicable: it could be made with a manual toggle setting, or use logic to ensure perimeter loops was greater than 1, or something along those lines.

On the cosmetic issue (which I hope it would improve): My thought is that as long as it was a feedrate move, it would not "rip" perimeters apart. If it is retreating into sparse infill, nozzle pressure would be allowed to bleed off into voids. Any scarring/blobbing there is unimportant in my eyes. 

Retreating into solid infill is a little more tricky, especially if it is a top surface. But assuming that perimeters are printed first, you could simply pause the extruder, and make a print move normal to the perimeter at that point, until the head is within the infill section, then begin printing infill. 

We could play around with tuning of that retreat print move to vary parameters like:

-How much extrusion on the retreat: no extrusion, or even a slight retract

-What speed the move should be (print speed, a bit slower / faster, etc). 

-Where should the retreat move begin? ie how close to the perimeter loop start point should we get before retreating?

I'm sure there is a combination that would minimize the surface scar, completely filling the seam without a bulge.

BagelOrb, 2015-08-10:
I think I misunderstood your initial request.

The feature should not have been disabled for a normal print, so I am thinking it is a bug that there is no move inward.

It's supposed to be a travel move, but indeed, perhaps a slight retraction will improve on that.

It's supposed to be travel speed right now, but I think somewhere in between print and travel speed could improve on that.

I think we should move further than the start point, thus performing a wipe of the nozzle on the start of the polygon.

You should also consider coasting. Coasting is a new feature which stops extrusion short of the end of a printed path and uses the oozed material to lay down the last bit of a line. This also reduces overextrusion at the end of a printed line.

I would like to start working on this, but at the time I am quite swamped with other features and bugfixes.
I hope I can process this feature request relatively soon.

CCS86, 2015-08-10:
Cool, I'm glad we are on the same page!

Is this coasting feature something new to Cura? I scanned back through the options and didn't see it listed. We have been talking a lot about extruder pressure management in some other threads (via "advance" in Marlin, Slic3r's "pressure advance", etc). 

I think it is an extremely important factor to control intelligently in a bowden printer.

BagelOrb, 2015-08-11:
Coasting is indeed a new feature. I think I took the naming from a similar feature in slic3r.

Can you please link to those threads? Are these threads on the forum?

CCS86, 2015-08-11:
https://github.com/MarlinFirmware/Marlin/issues/916

https://ultimaker.com/en/marlin/view/6693-blobless-printing-in-marlin?page=1

BagelOrb, 2015-08-17:
`moveInside` lives once more!
I didn't really feel the need to make the distance configurable. Now it moves to a point at [nozzle_size] away from the comb boundary, which is generally the second perimeter.

closed 2015-08-17 by BagelOrb

CCS86, 2015-11-30:
@BagelOrb 

Have you verified that this is still working?

I made a print last night, and was having trouble with long combing moves oozing too much material. So I disabled combing, which obviously forced more retract moves. This ended up making warts all over the outer perimeters, when the head dwelled right on the outer to retract and prime.

BagelOrb, 2015-12-01:
I will have another look at it. 

This is on my TODO list.

BagelOrb, 2015-12-02:
Indeed it was the combing code which also performed the move inside.

I'll see if I can fix this.

BagelOrb, 2015-12-02:
Fixed for the next release!

CCS86, 2015-12-02:
Thanks @BagelOrb !

Any idea when the next build will be posted?

BagelOrb, 2015-12-02:
Could be a couple of weeks, could be a couple of months. My guess would be a month or so, but don't count on it.

CCS86, 2015-12-02:
Oh, shoot. I guess I got spoiled by three releases in the first half of Nov

BagelOrb, 2015-12-02:
Oh yeah a beta release might come sooner. I don't really have a perspective
on those.

Ghostkeeper, 2015-12-03:
The idea is to make a weekly build. It's been cancelled last time due to blocking bugs though, which made the build unusable.