Ultimaker/CuraEngine #229: 15.06 gcode does not overlap mesh edges 15.04 does

ThatJCB, 2015-08-09:
I spent all my Sunday morning trying to print a model which just fell apart during printing each time. 

The print was not overlapping anywhere.  The mesh is made up of separate closed manifold objects,  as I have done with most of my more complex models.  Those objects fully intersect in blender and the stl file. 

With the gcode from 15.06.03 the outer edges never intersect,  they barely touch.

Eventually I reverted to 15.04 and it is now printing as expected! 

In my opinion this is a big issue with 15.06 if it is not as good at printing as 15.04

ThatJCB, 2015-08-09:
I've posted some photos of the problem on my blog:
http://blog.discoverthat.co.uk/2015/08/trouble-printing.html

BagelOrb, 2015-08-10:
I've read the blog but I still don't quite understand what the problem is.

Specifically, can you provide me with info on:
- Steps to reproduce the bug. Can you provide me with the test STLs of A and B?
- What you expected to happen.
- What actually happened.
- What (relevant) settings did you change from their defaults? Did you make some (relevant) settings visible?

daid, 2015-08-10:
@BagelOrb this is the remove overlapping segments change that you did that's causing parts of the print to disappear.

BagelOrb, 2015-08-10:
@daid Thanks! Now I understand.

The tests you did are not directly related to your problem.

The problem your model has is that it has too thin walls - that is: there are some places where the width of the model is less than 0.8 (twice the nozzle diameter). In 15.06 these pieces are automatically removed, because the option 'Remove Overlapping Wall Parts' defaults to 'True' instead of 'False'.

In the next release it will default to 'False' and there will be an option 'Compensate Wall Overlaps' which instead of removing such parts adjust the flow when parts are between 0.4 and 0.8 mm thick.

The tests you posted are probably walls which are thinner than 0.8 mm and so laying down a perimeter on each side of the wall would result in overlapping perimeters, which is a form of overextrusion.

closed 2015-08-10 by BagelOrb

ThatJCB, 2015-08-10:
Thank you, it is the overlapping parts that have problems.

I can't find the option 'Remove Overlapping Wall Parts'
Where is it?

There is an option, 'Avoid Overlapping Walls' but that appears to default to disabled.

The sample I used deliberately has thin walls as it is the quickest way to demonstrate the problem. I also want to be able to print thin walls in order to create my own supports so having walls of only 0.4mm should be a valid part of a design.

CCS86, 2015-08-10:
I have found that enabling "avoid overlapping walls" provides much better prints by filling in small areas that would otherwise have been left as voids.

BagelOrb, 2015-08-10:
@ThatJCB Indeed the setting has its proper name in the next release, That's why it's different in the current version.

You shouldn't create models with parts smaller than 0.8. This has never been processed properly in Cura - it lead to massive overextrusion. However, in the next release it will be possible.

In the next release it will probably be possible to provide support as surface meshes instead of volume meshes. A single vertical polygon of a mesh would then be printed as a vertical wall of single lines.

I hope these developments are to your liking. 

As of yet I cannot reproduce your bug. Please provide the requested information.

@CCS86 Actually it is not "avoid overlapping walls" which provides the functionality but "Fill Gaps Between Walls" (at least that's what it's called in the next release), which works at more places when "avoid overlapping walls" is enabled. The removed parts are then filled afterwards.

ThatJCB, 2015-08-10:
The test file, as you have said, has walls smaller than 0.8 so may produce the symptoms for another reason.

The significant problem is the walls not joining and having gaps between them.  The model that does that is discovery body.  

Download temporarily from here:
https://drive.google.com/file/d/0B6KA9qkK3ar5eEYzRDh4b3kyOG8/view?usp=sharing

There are no steps to reproduce the problem.
- Drag the stl file in to Cura 15.06.03
- Set just standard settings in advanced mode
- Save to disk
- Copy to SD card
- Take to the printer and print

All the settings I changed are shown on the screenshots in the blog:  
Layer height 0.12mm
Shell thickness 0.8mm
Enable retraction
Print speed 40mm/s
Travel speed 150mm/s
Infill density 18% (although I used 22% when I reprinted in 15.04)
Brim enabled
Brim line count 20
When I reprinted I used the built in support only from the bed.

What I expected to happen was that where I had overlapping meshes they would merge in to one another, ignoring any hidden geometry and following the outer edge of the combined shape.  Exactly as it does by default in 15.04.

What happened is that it appears to have done a Boolean intersect and removed the intersection leaving two separate objects that only just touch at the corners.  The overlapping inner geometry is completely hollow! The print head follows the outline of each separated shape so no plastic links them together.

It's very hard to describe but I hope that helps.

Thanks