Ultimaker/CuraEngine #231: Cura 15.06.02 does not create infill in small shaft

3Dnerd, 2015-08-10:
Hi,

today I wanted to print a small holder.
I sliced it with new Cura 15.06.02 and watched the layers first.
But new Cura does not create any infill inside the shaft (around layer 44).
In the bottom and top it creates infill again.
Then I sliced it with legacy Cura 15.02.01 and this one creates the infill without any problems as you can see here:

http://i.imgur.com/6h6x0rG.jpg

I posted this problem also on the regular ultimaker forum:

https://ultimaker.com/en/cura/view/16787-cura-150602-does-not-create-infill-in-shaft?page=1

gr5 confirmed this bug!

Here is link the link to the STL-file:

https://www.dropbox.com/s/dvgna1cx648hfrw/Holder.stl?dl=0 

Thanks!

BagelOrb, 2015-08-27:
Sorry for the delayed response.

Small areas of infill are intentionally removed. It seems that if your model was placed a bit to the left the part would have missed the infill anyway.

Did you experience any problems when printing this?

BagelOrb, 2015-08-27:
Oh Im sorry, I missed the 50% infill part.

Note that you can insert images just by dragging them into the text field where you're typing.
[image]

BagelOrb, 2015-08-27:
Ok, thinking about it your area is really too big to be not filled with infill.

This is fixed for the next release.

closed 2015-08-27 by BagelOrb