Ultimaker/CuraEngine #232: Ultimaker Original refuses to print with the newest Cura 15.06.03

TheNeuron, 2015-08-11:
Recently I updated Cura to 15.06.03 and updated the firmware of the Ultimaker, now Cura refuses to print since it wants to "Bed heating" even though there is none installed. The printer is defined as an Ultimaker Original and every box concerning the bed heating is disabled. Still in the display of the Ultimaker it says "Bed heating". Since there is no Bed to heat it does not print.

awhiemstra suggested to put it here, there is a workaround in enabling bed heating and set it to 0.

awhiemstra, 2015-08-11:
Basically, the issue is that the engine should check if the setting "machine_heated_bed" is true before adding the bed heating code.

BagelOrb, 2015-08-11:
It should be doing that already. This is a bug I have to look into..

BagelOrb, 2015-08-24:
This should be fixed in the next release.

closed 2015-08-24 by BagelOrb