Ultimaker/CuraEngine #233: Blemishes on smooth surface

CCS86, 2015-08-16:
Any ideas what is going on here? The model surface looks clean, and Slic3r makes a smooth surface:

15.06.3
[image]

Slic3r:
[image]

Cura Gcode: https://dl.dropboxusercontent.com/u/60958586/directdrive.gcode

Model: https://dl.dropboxusercontent.com/u/60958586/Direct%20Drive%20Bracket.stl

CCS86, 2015-08-16:
This is two sides of the same print from Cura Gcode:

[image]

[image]

[image]

CCS86, 2015-08-16:
It's odd how the visualization defects look a fair bit different than the print defects.

BagelOrb, 2015-08-17:
What kind of printer do you have? I.e. what machine do you select in Cura?

What settings did you use?

BagelOrb, 2015-08-17:
I can't reproduce your bug. The gcode you sent looks just fine in the old Cura and Repetier-Host.

Are you sure you included the correct gcode?

What program are you using to display the gcode?

CCS86, 2015-08-17:
I have an UMO. What settings are you interested in knowing?

I am visualizing in Repetier. Are you sure you just don't have too many
lights turned on? That can wash out the blemishes. Also, which
visualization method are you using? I am using the width/height ratio and
filament diameter.

BagelOrb, 2015-08-17:
Please re-upload the correct gcode.

I am viewing in Repetier, as I am fairly certain you accidentally uploaded the slic3r gcode or something.

from which angle are you viewing the gcode? I deduced it was diagonally from the side/top above the left bulge on the bottom with the hole in it. Is this correct?

[image]

CCS86, 2015-08-17:
I think you do not have visualization set the same:

[image]

Doing it this way not only catches positional issues, but extrusion distance relative to print move distance issues.

[image]

CCS86, 2015-08-17:
Same area set to "layer height":

[image]

BagelOrb, 2015-08-17:
Ah yes.

Well on first view I notice that the width/diameter of the extrusion paths is conical instead of cylindrical, while firmware interprets every move linear, and so the correct vizualisation is cylindrical. Perhaps Repetier is using some heuristic which generally leads to better vizualisation when the extruded volume gradually increases over a couple of moves, but in this case it doesn't seem to perform so well.

The old Cura also vizualizes the width per extrusion move, but doesn't make them trapezoid; it leaves them square. In the Cura gcode view no such blemishes show up.

However, the gcode portrayed seems to overextrude by 100%. Didn't you maybe accidentally set the flow to 200%?

What version of Cura are you using and which settings have you changed from their defaults?

Could you maybe upload your settings file?

CCS86, 2015-08-17:
My settings change print to print, so I don't have those exact settings. I thought that they were being written to the gcode, but it appears not. That would be pretty helpful for diagnostics. If you ask me about specific settings, I'm sure I could tell you.

I listed the version in the first post: 15.06.3

How could I have gotten an otherwise great print, if I had set my flow to 200%? I don't change flow rate in the slicer.

You say that Repetier is not doing a good job with visualization, but I disagree. It shows that the surface will be blemished, where Cura shows it will be fine. Repetier is clearly more accurate here. When I visualize the same model sliced with slic3r, it shows none of these extrusion based defects.

We can debate semantics of gcode visualization all day, but the print is what matters. The print looks terrible in some places. To me, it looks like exactly what Repetier is showing: Massive segments of over-extrusion on the outer perimeter are "casting" filament out in loops. The defects looks much smaller in visualization than in real life, because the extruder pressure has a bit of "inertia" that isn't modeled in the software.

BagelOrb, 2015-08-17:
As to how well Repetier vizualizes: you yourself said "It's odd how the visualization defects look a fair bit different than the print defects." I didn't say Repetier is doing a bad job, though.

I think the Repetier vizualization is perpetuating the zigzag endings of the infill toward the outer perimeter.

It seems you are using 100% infill. I think it is the infill overlap which causes the overextrusion. Did you set the infill overlap to 0%?

I changed the default inherit structure. Now when infill is above 95%, the infill overlap is disabled.

I trust this solves your problem -- otherwise I will reopen this issue.

CCS86, 2015-08-17:
That is a huge assumption, and not correct.

If this was a global issue, like bad infill overlap amount, it would show
up everywhere. With a shell thickness of 4 passes, there is no way that
infill overlap could cause these blemishes.

BagelOrb, 2015-08-17:
Well, I don't know what to do any more. 

We both cannot reproduce the bug - I because I don't have the settings and you cannot because you have already changed the settings.

Moreover we have no way to verify there even is a bug.

Just to clarify on this last point: Repetier views conically shaped extrusion paths, while these are impossible -- gcode cannot specify a conical path. The blemishes showed in the view seem to be a result only of the conical shape of the extrusion paths.

CCS86, 2015-08-17:
I don't understand why you are hung up on the conical shape. The important aspect here is that the Cura generated code has blatant over extrusion. This is visible in the visualization, and in the actual print. How much more reproducible can it be?

If I visualize code generated by Slic3r (with the same visualization settings), it shows no sign of over-extrusion.

BagelOrb, 2015-08-17:
I don't understand. How do you see there is over extrusion? 

The reason the visualisation of the slic3r gcode is different because the infill has a zigzag pattern instead of a lines pattern.

I cannot reproduce the gcode because I don't have the settings. Please provide some settings which likely produce the problem.

The fact that I cannot reproduce the gcode makes it very difficult for me to determine what's going on. I cannot verify whether the massive over extrusion I see in cura is a visualisation issue.

Please try to find which settings lead to the problem.

CCS86, 2015-08-17:
The reason visualization looks different on the Slic3r generated code has nothing to do with infill pattern. The issue here is the extrusion amounts on the perimeters. Period.

"I don't understand. How do you see there is over extrusion?"

I don't understand your confusion. Both the visualization and the actual print show massive over-extrusion, but only in finite section of the code. Please go back and look at the actual photos I posted. Both were from the same print, with Cura generated code. Some portions of the exterior look flawless, others are terrible. Printer settings remained constant throughout the print. Bad code is to blame.

I had no fancy settings in Cura for this. 

.2mm layers, 100% infill. I don't have time to test every single possible settings combination to see if there is one, where Cura works correctly. For the third time though, if you have questions about specific settings that you think could contribute, ask me.

Throwing your hands up and pretending that this is not a bug, or is too hard to reproduce, certainly isn't going to help fix it.

BagelOrb, 2015-08-17:
Your tone of voice is not suited. I'm not pretending this is not a bug. You can restate your claim on the visualisation, but that has little use with neither of us testing or verifying anything.

I don't know beforehand which settings influence it; that is why I'm asking for a more complete picture. 

I will test print the mesh with .2mm layer height and 100% infill on my UM2 testing machine.

I'll get back to you when it's done.

CCS86, 2015-08-17:
Closing the ticket before the issue is sufficiently discussed implies that you don't think it is a bug.

I'm happy to discuss and try to help, but getting stonewalled and having the issue closed almost immediately is annoying.

BagelOrb, 2015-08-17:
I closed the issue because I have no means to solve it. I don't even have proof it is a software issue.

There is no further discussion needed. I need to reproduce the problem. If I cannot reproduce it, I cannot solve it.

BagelOrb, 2015-08-17:
Just to be clear: I will try to reproduce it.

CCS86, 2015-08-17:
I tried to analyze some of that code by hand to see what is going on, and actually found evidence that the visualization issue is independent of bad extrusion amounts. I looked at one section of code that involved Y travel, no real X movement, and showed visualization artifacts. I calculated the delta Y and delta E. Using a simplified model of extruded volume (.2mm * .4mm * length) and a straightforward volume calc for extrusion volume, I found the averages to match well:

G1 X172.003 Y145.346 E641.89972
...
G1 X172.002 Y151.894 E641.98185

Y travel = 6.548 mm

E travel = 0.08213 mm

Filament volume (L * W * H)= 0.52384 mm^3

Extruded volume (Pi * R^2 * L)= 0.52393 mm^3

Then I looked at the differences between these different segments:
[image]
[image]
[image]

I found them to have almost identical extrusion mm / print head mm amounts as well. Why repetier shows the odd shapes, I do not know! We can forget that aspect, since this thread isn't about Repetier.

I am curious why cura is breaking the print moves up like this. But more than anything I want to prevent ugly prints like I got, and get to the root cause.

BagelOrb, 2015-08-17:
Against overextrusion I would advise against 100% infill.

As to the breaking up into small parts: isn't it true the shape is a little twisted? Isn't the line just not a perfect straight long line, but very slightly curved?

CCS86, 2015-08-17:
Seriously, closed again?

The surface in question is not curved or twisted. even looking at the Gcode, X bounces back and forth +/- 0.001mm. I understand that there are reasons to break up segments, but why these 0.01mm long segments?

The original issue remains: The print has sections which are nearly perfect, and section with "loops" of filament cast out beyond the finished surface.

I don't know why you are so stuck on 100% infill being a bad idea. It is not causing this issue, nor is it a bad idea. By definition, the perimeter area of a print is always 100% infill, and that is the problem area. For maximum strength parts, I'm going with 100% infill.

BagelOrb, 2015-08-17:
You keep claiming to know things better and it takes quite long to convince you. Now that you have convinced yourself of the visualization problem, how long will it take you to realize the surface is curved?

You should try to listen instead of reassuring yourself with air castles.

CCS86, 2015-08-17:
Hahaha.

I have no problem admitting when I am wrong, which is why I continued digging into this issue and posted my findings. If you want to take the "opportunity" to say I told you so, and it makes you feel better, by all means.

I don't "keep claiming to know things better". Both you and I made assumptions earlier. I assumed that the visualization blemishes, which coincided with actual print blemishes, were related (not a huge stretch). You assumed Cura's Gcode was flawless. As far as I am aware, you didn't do the math on the Gcode I uploaded. How you feel this gives you high ground, I don't know.

This whole 100% infill thing is a different matter altogether. I have printed hundreds and hundreds of prints with 100% infill and not had any problems. That is more than enough empirical data to make the statements I have. What "problems" are you even talking about? If the infill regions are small, and have many reversals, and your acceleration/jerk is set low, you will run into some over-extrusion in these areas (until Marlin implements pressure management anyway). Not a big deal. I'd rather have a touch of over-extrusion than all my infill sparse and not connected.

Whether the surface is curved or not, is a complete non-issue as well. That surface was formed buy lofting 2 sketches. Both sketches have lines in that region. There might be a bit of overhang, but any 2 dimensional slice should be a line in that area. Again, it doesn't even matter. Why would Cura generate segments 0.01mm long?

Here is a section of that Gcode from layer 120 in that region:

G1 X172.002 Y145.029 E641.89575
G1 X172.003 Y145.346 E641.89972
G1 X172.002 Y145.356 E641.89985
G1 X172.003 Y145.673 E641.90382
G1 X172.002 Y145.683 E641.90395
G1 X172.003 Y145.990 E641.90780
G1 X172.002 Y146.336 E641.91214
G1 X172.003 Y146.653 E641.91611
G1 X172.002 Y146.663 E641.91624
G1 X172.003 Y146.980 E641.92021
G1 X172.002 Y146.990 E641.92034
G1 X172.003 Y147.307 E641.92432
G1 X172.002 Y147.317 E641.92444
G1 X172.003 Y147.634 E641.92842
G1 X172.002 Y147.644 E641.92854
G1 X172.003 Y147.961 E641.93252
G1 X172.002 Y147.971 E641.93264
G1 X172.003 Y148.288 E641.93662
G1 X172.002 Y148.298 E641.93675
G1 X172.003 Y148.615 E641.94072
G1 X172.002 Y148.625 E641.94085
G1 X172.003 Y148.942 E641.94482
G1 X172.002 Y148.952 E641.94495
G1 X172.003 Y149.269 E641.94892
G1 X172.002 Y149.279 E641.94905
G1 X172.003 Y149.596 E641.95303
G1 X172.002 Y149.606 E641.95315
G1 X172.003 Y149.923 E641.95713
G1 X172.002 Y149.933 E641.95725
G1 X172.003 Y150.250 E641.96123
G1 X172.002 Y150.260 E641.96135
G1 X172.003 Y150.577 E641.96533
G1 X172.002 Y150.587 E641.96546
G1 X172.003 Y150.903 E641.96942
G1 X172.002 Y150.913 E641.96954
G1 X172.003 Y151.230 E641.97352
G1 X172.002 Y151.240 E641.97365
G1 X172.003 Y151.557 E641.97762
G1 X172.002 Y151.567 E641.97775
G1 X172.003 Y151.884 E641.98172
G1 X172.002 Y151.894 E641.98185
G1 X172.003 Y152.211 E641.98582
G1 X172.002 Y152.221 E641.98595
G1 X172.003 Y152.531 E641.98984
G1 X172.002 Y152.864 E641.99401
G1 X172.003 Y153.186 E641.99805
G1 X172.002 Y153.519 E642.00223
G1 X172.003 Y153.840 E642.00625
G1 X172.002 Y154.173 E642.01043
G1 X172.003 Y154.494 E642.01445
G1 X172.002 Y154.827 E642.01863
G1 X172.003 Y155.148 E642.02266
G1 X172.002 Y155.481 E642.02683
G1 X172.003 Y155.802 E642.03086

Does that look like a curve to you?

BagelOrb, 2015-08-18:
I told you that the visualisation was off and you didn't listen. Instead of accepting my proposition or disapproving it you just claimed the opposite. This is not a discussion however; this is a github issue. I am helping you with my expertise. Therefore you plainly ignoring my claims and claiming the opposite is rude.

Cura generates those segments because that's what the model looks like. However, cura should optimise away aligned segments. It's not a bug it's a missing feature ;)

CCS86, 2015-08-18:
False. 

Here is what I said:
"You say that Repetier is not doing a good job with visualization, but I disagree. It shows that the surface will be blemished, where Cura shows it will be fine. Repetier is clearly more accurate here. When I visualize the same model sliced with slic3r, it shows none of these extrusion based defects.

We can debate semantics of gcode visualization all day, but the print is what matters. The print looks terrible in some places. To me, it looks like exactly what Repetier is showing: Massive segments of over-extrusion on the outer perimeter are "casting" filament out in loops. The defects looks much smaller in visualization than in real life, because the extruder pressure has a bit of "inertia" that isn't modeled in the software."

At that point in time, I disagreed with you on the visualization aspect (is that really a big deal? Will you ever get over it?), but continued to spend time getting to the truth of it. I proved you right on that aspect, and you continue to lord it around like you won a medal. How is that "ignoring your claims", as you say?

Even at that point in the conversation, I was trying to downplay the visualization aspect, and focus on the important part: The print looks terrible. That has not changed, it still looks terrible. You don't seem interested in helping figure out why. You said you would print the model and report back. Did you?

You claim Cura generates these 0.01mm long segments because that is "what the model looks like".  Can you back up that claim? Here is the model:

[image]

I see nothing but uniform face sizes in that region.

You claim that the surface is curved (not that it matters), and I show you evidence to the contrary. You ignore it and focus on your "visualization victory".

I ask you to explain why you think 100% infill is bad, or to blame. You go quiet.

I am willing to admit when I am wrong and search for evidence without bias. I speak to your points and try to have a productive discussion. You make claims without any willingness to back them up. You avoid acknowledging any evidence that doesn't support your claims. How is this "helping me"?

I love Cura, and want to help make it better, but I am not a Cure developer, so I can't do that alone. I need willingness on your part. 

Can we get this back on track?

nophead, 2015-08-18:
Your model is tessellated in an unusual way. It seems to have uniform sized triangles regardless of curvature. That means the slice will have lots of short segments, even on flat surfaces. Possibly that simply exceeds the comms bandwidth and your printer stutters and pauses. During the pause the flow will not stop instantly, hence the loops.

Other slicers can combine sort segments to avoid comms overload.

BagelOrb, 2015-08-18:
The model contains a horizontal line of triangles; just below such a line, all triangles are sliced by just the tip, which is very small when the slice is very near to such a line.

Indeed the surface is not curved at that point. However, the triangles *are* part of the model and thus so are the small segments.

You haven't asked me to explain why I advise against 100% infill, but it is implied in your last post. Because of many heuristics and simplifications, the infill area is not perfectly filled. For example the infill lines are modelled as square, while the infill area may have round corners. These simplifications can cause over or under extrusion in certain areas. When using less than 100% infill overextruded filament can escape inside. Otherwise it will build up and finally affect the outer surface. However, I suspect this effect to be quite negligible. 

I have made claims without backing them up. I've also made claims which I *have* backed up, but which you seemed to ignore. I've also ignored some of your true claims to which I should have said that you were right. Now let's end this feud.

Just repeating myself to end this issue on-topic:
**The bug that still has to be solved is a bug in the optimization which reduces the polygons size by combining similarly aligned segments.**

CCS86, 2015-08-18:
Yes, I dictated that during the export of my model. I have a deviation tolerance and max facet size. I set the max facet size to 1mm in hopes that it would actually help the slicer. My thought being that since faces get sliced into lines and then usually the endpoints are used by the slicer. By keeping the faces small, it puts endpoints closer to wherever the print head is. 

I see now how at certain heights, you end up with alternating line segment lengths, with some being tiny.

The idea about the printer "choking" on these short moves is interesting. I didn't watch the print to say for sure, but the "loops" on the print are 2-3mm long, which isn't what I would expect from brief dwells. 

Did you get a chance to print that Gcode @BagelOrb ?

BagelOrb, 2015-08-19:
Yes. Note that I didn't print your gcode (since I was testing on an UM2), but instead sliced the mesh again using similar settings.

The surface in question looks like it has scales, rather than smooth. I'm not getting the specific results you had gotten.

(Note that the warts are due to some unrelated bugs)
[image]
[image]

CCS86, 2015-08-19:
Thanks @BagelOrb !

That is very interesting. The results aren't exactly the same, but very similar on that surface to what I got. You seem to have 3 columns of those "scales", where as mine were a little more disorderly. But the shape of each scale looks almost the same!

Di you happen to watch the machine while it printed those? I'm very curious if that is an anomaly in the motion which is the cause.

BagelOrb, 2015-08-20:
Unfortunately I didn't watch the printer as it went. I was too busy trying to solve the bug ;)

BagelOrb, 2015-08-20:
[image]
Solved!

closed 2015-08-20 by BagelOrb

CCS86, 2015-08-20:
Well done sir!

Now I am super curious to see if that resolves the "loop" blemishes.

BagelOrb, 2015-08-20:
Note that I just solved the warts bug as well, so I am printing the thing once more! ^^
[image]

BagelOrb, 2015-08-20:
Also: looks like I need to perform the atomic method...

nickthetait, 2015-08-20:
Awesome fix!

CCS86, 2015-08-20:
Sweet! 

On the plus side, you can convert some UMO's to direct drive now ;)

Can I ask what the atomic method is?

nickthetait, 2015-08-20:
I don't work for Ultimaker, but here's the atomic method. A very popular way to clean out a nozzle. https://ultimaker.com/en/support/view/149-atomic-method

CCS86, 2015-08-20:
Ahh, thanks!

BagelOrb, 2015-08-24:
It's just downloadable at software.ultimaker.com ...
I would say it's open beta right now, although it's being perceived as a release.

The warts bug is also solved now (https://github.com/Ultimaker/CuraEngine/issues/228)
[image]

CCS86, 2015-08-24:
Looking good!

Any idea of when a new build will be released?

BagelOrb, 2015-08-24:
It'll be around september/oktober.

CCS86, 2015-08-25:
Can't wait to get my hands on it!

nallath, 2015-08-25:
The (open) beta is planned for 1 september.