Ultimaker/CuraEngine #234: Problem using CuraEngine downloaded from Cura

diridiri, 2015-08-17:
Hi Ultimaker,
Recently I've downloaded Cura from ultimaker.com (version 15.06.03) and extracted CuraEngine.exe from the directory.
I've prepared dlls and other files needed for CuraEngine, and I tried to slice stl file on command line (win64 cmd) as "CuraEngine.exe slice -v -j asdf.json -o 'output.gcode' -e0 -l 'asdf.stl'". 
But result returned was Error "Failed to load model: slice". 
Is there something wrong with my command or do i need to compile it in windows?(actually i don't know the way in windows :( )

BagelOrb, 2015-08-17:
You have compiled the 15.06 branch, while you are calling the command line using a structure which is new and only possible for the master branch.

You may choose:
- switch to the master branch and compile that
- call it with `CuraEngine.exe -v -j asdf.json -o 'output.gcode' 'asdf.stl'`

closed 2015-08-17 by BagelOrb

diridiri, 2015-08-19:
thanks a lot for your answer it worked

BagelOrb, 2015-08-19:
Just for my interest: why do you want to slice via the command line?