Ultimaker/CuraEngine #236: Polygons in json are not recognised

nallath, 2015-08-21:
The FDMPrinter.json specifies a number of polygons, which are not recognised by the engine.

Error log;
Unrecognized data type in JSON: machine_disallowed_areas has type 4
Unrecognized data type in JSON: machine_platform_offset has type 4
Unrecognized data type in JSON: machine_head_polygon has type 4
Unrecognized data type in JSON: machine_head_with_fans_polygon has type 4

BagelOrb, 2015-08-24:
Indeed, they are still only used by the frontend.

The problem is that the backend doesn't know what to do with an array yet.

BagelOrb, 2015-08-24:
In general a setting value is a single string. The polygons break that rule.

I think for now it's okay to have this unimplemented.

Should I remove the error notice?

nallath, 2015-08-24:
Yeah, but only for polygon arrays. Other not recognised settings should give an error.

BagelOrb, 2015-08-24:
Then I would like to see some "type": "polygon" in the JSON.
I will do so.

closed 2015-08-24 by BagelOrb