Ultimaker/CuraEngine #237: Feature request: Support Interface Density

CCS86, 2015-08-24:
Hi guys, any chance we could have a feature similar to Slic3r where there are additional settings for support pattern spacing / density for the final layer(s) of support? I find it very useful in keeping support fast and easy to remove, but allow a more dense platform for the print to recover more quickly.

Thanks!

BagelOrb, 2015-08-24:
Currently (on the master branch) a hammock is possible. This is a fully dense interface layer with configurable thickness (number of layers).

Would you like to configure such a hammock further?

CCS86, 2015-08-24:
I think it would be far more useful if the density could be configured as well, either through a percentage, or just a line spacing parameter. It would really help tune the ease of detachment.

BagelOrb, 2015-08-25:
Should it then also have a configurable pattern (ZigZag, lines, concentric)?

CCS86, 2015-08-25:
I don't think it would hurt. I haven't played much with different interface
patterns.

BagelOrb, 2015-08-25:
[Implemented]

closed 2015-08-25 by BagelOrb