Ultimaker/CuraEngine #240: Unable to choose heat order or wait for temperature in start gcod

markwal, 2015-09-06:
For reprap flavor gcode:

I believe (see issue #67) that there used to be a way to have tokens in start gcode plus if the start gcode contained the set temperatures then curaengine wouldn't set the temperatures itself.  Now, however, a machine profile with "machine_start_gcode" doesn't appear to get token replacement and cura engine insists on emitting first wait for bed then wait for extruder temperature if "material_bed_temperature" is a setting that is shown.

So, if you'd like to be able to specify the temperature in the UI and have the heat up be simultaneous and after some portion of the gcode, I don't believe there is an easy way to do that.

BagelOrb, 2015-09-08:
I find it highly unlikely the start_gcode was ever parsed. @daid ?

This feature request seems low priority. If you work with a custom start_gcode you could also customize it such that you change the temperature, right?

markwal, 2015-09-08:
Yes, but I'd like to choose the temperature setting in the UI and not choose a different printer profile for every temperature I might want to try.  I could rig up a PR that does these two things:

1. simple token replacement on the start/end gcode (just straight pattern substitution, no parsing)
2. if any of the following exist (simple text match ignoring whether they are in comments or not) in the start gcode then the built-in temperatures are not emitted at the beginning: M104, M140, M109, M190

markwal, 2015-09-08:
It wasn't in the engine before it was in Cura, but so was the M190 emit:
https://github.com/daid/Cura/blob/9b5aa5c182a6c2285f1f5f362ac0eeb31d3ff222/Cura/util/profile.py#L1320

BagelOrb, 2015-09-09:
I think this should be done in the new frontend as well.

It would then be possible to do token replacements in each gcode setting.

Where the string contains "{...}" the text should be replaced by the setting which named within the braces.

markwal, 2015-09-09:
Ok, but there is still the problem that the engine now emits the M190 & M109. Should that move to the front end as well?
Would a pull request be considered and how is a change done across both the back end and front end?

BagelOrb, 2015-09-09:
A pull request would certainly be welcome!

You are right to say that the first heat up gcode commands generated by the engine should be suppressed (in the engine) if the start gcode already contains the heat up commands. However, what would you do if the temperature in the start gcode doesn't coincide with the generated temperature command? I would say the generated temperature command should 'win', since it's more likely that that temperature has been tweaked by the user.

The frontend would still have to replace the `{temperature}` by the actual temperature.

A change over both the frontend and backend is generally performed by pushing both repositories at the same time. We can just merge your pull requests on both repositories at the same time.

markwal, 2015-09-09:
Today the generated temperature doesn't win since it is emitted first. Well, it'll win in causing the wait, but not in choosing the temperature.

BagelOrb, 2015-09-10:
Indeed. Are you working on a pull request?

markwal, 2015-09-11:
I haven't started yet, but I did move my Cura repo's from 16.03 to latest master and got my build working again.  OK, if I understand correctly, the idea would be:

* move the current logic that is in Wireframe2gcode.cpp and FffGcodeWriter.cpp that looks for material_bed_temperature, material_print_temperature from writer::processStartingGcode to CuraEngineBackend.py
* Add a new function to CuraEngineBackend (def setGcodeScripts()) that gets called from slice before _sendSettings (or in _sendSettings?) that does the token replacement of all settings tokens into the start and end scripts.

Probably the thing to do is to do the token replacement on the way out on _sendSettings and leave the value in the profile unmodified.

markwal, 2015-10-17:
@BagelOrb OK.  Getting around to this finally.  I've got it so that it replaces the placeholders that correspond to settings that are in the profile that is sent to the backend via sendSettings.

Conditionally suppressing the automatically emitted heat commands is harder.  My first thought was to move them entirely out of FffCodeWriter::processStartingCode and put them into machine_start_gcode in sendSettings.  The reason this won't work is that the current code uses a per mesh setting for the nozzle temperatures.

As an aside, It's a bit nonsensical if the meshes are using the same nozzle since randomly the last one will win.

But in the case where one mesh is from one nozzle and the other mesh from another nozzle, it makes some sense that they'd have different temperatures.  Also, this is the way that only the used nozzles are set to temp.   I'm worried that sendSettings may be the wrong time to do the token replacement (though I guess it has to run again every time the engine is restarted due to settings change, right?)

But worse, it means the old Cura's token replacement isn't quite sufficient since the new Cura may handle more than one nozzle.  I propose some new tokens:

{set_bed_temperature}
{set_bed_temperature_and_wait}
{set_all_extruder_temperatures}
{set_all_extruder_temperatures_and_wait}

If they exist in the start gcode, I'll use the loop and emit all the commands there and suppress them from before the start gcode.  Otherwise, I'll act as if they appear at the very beginning (giving the current result).  What do you think?

Also, noob question: is storage.meshes accessible somehow from the CuraEngineBackend.py?

BagelOrb, 2015-10-19:
storage.meshes is not accessible from the frontend. The frontend has it's own way to represent the per object settings.
Note that in the engine the per object settings inherit from the extruder trains which inherit from the global settings. So getting a temperature setting from an object most likely comes down to getting it from the corresponding extruder train.

markwal, 2015-11-24:
Fixed by having placeholders in start/end gcode

closed 2015-11-24 by markwal