Ultimaker/CuraEngine #241: Infill combine: skip every x layers

BagelOrb, 2015-09-10:
now 3-layer infill is somehow computed per layer
[image]
[image]

I would expect it to be computed every 3 layers, so that all 3-layer infill lines are in every third layer and not in layers in between.

CCS86, 2015-09-30:
I'm having no luck with the combine infill setting in 15.09.85. Seems to just generate infill every layer when using "infill layers" set to 2

BagelOrb, 2015-09-30:
Perhaps it's not viewing the combined layers. try to export to gcode and load it into the old Cura.

It seems to work at least on the latest version.

BagelOrb, 2015-09-30:
No the view is also correct...

CCS86, 2015-09-30:
Tried that.

BagelOrb, 2015-10-01:
It's really strange it doesn't seem to work at all for you. 
I haven't had any such problems on any 15.10 pre-release...

What version are you testing on right now?
What model are you testing on?

CCS86, 2015-10-03:
I am on 15.09.85 now. I just tried again on a simpler model, and it was working as expected.

But I still can't get it working with that starfish lampshade model. I scaled it down to 50% and disabled support for faster generation. With infill layers set to 2 and 35% density, I am getting infill every layer.

closed 2015-10-09 by Ghostkeeper