Ultimaker/CuraEngine #248: [master] Coasting causes freeze

Ghostkeeper, 2015-09-24:
When coasting is enabled, slicing freezes on my model at 42%. I can't easily test this on the release candidate since the installer won't work, but I understood the engine is more or less the same.

Version: Cura master commit fdd485a884daaaacb170471d502048fc9b6c0d83, CuraEngine master commit e861fba11b7f3e12d8930886848e5ea6f0d00796 (most recent master versions as of September 24).
Platform: Ubuntu 15.04 Unity
Qt version: 5.4.1
PyQt version: 5.4.1

Steps to reproduce:

* Download [this](http://dulek.net/persoonlijk/coathook.stl) model.
* Load it into Cura with the default settings (normal profile).
* Enable coasting. Watch it slice.

Expected results:

* CuraEngine slices the model. The progress bar completes its course to 100%.

Observed results:

* CuraEngine starts slicing, but at 42% the progress bar gets stuck and never continues.

Ghostkeeper, 2015-09-24:
I have since confirmed that this problem also occurs in branch 15.10.

BagelOrb, 2015-09-24:
I was thinking of removing the feature altogether. It introduces another
level of complexity and hasn't produced any good results on any settings I
have tested.

Still up for discussion though...

Ghostkeeper, 2015-09-24:
I've seen @TMHogenhout and @BagelOrb use coasting without problems, so it's not occurring on every object.

CCS86, 2015-09-24:
I have also used it without any crashes. It has a few small glitches in the
code, but nothing too bad.

I've been meaning to test more thoroughly and try tuning the coast
parameters.

I'm not sure I understand the current implementation. Does it just work
backwards from the print move endpoint, by the distance needed to hit the
coast volume parameter, then run that small segment as a travel move, at a
fixed speed defined by the coast speed multiplier and the original print
move speed. Is that right?

To me it seems like instead of a fixed speed, that segment should be run in
constant deceleration, since extruder pressure is bleeding off, and flow
with it.

BagelOrb, 2015-09-25:
@CCS86 that's about right.

Coasting is performed not only to counter the pressure decay; it's also
used because there is a small amount of plastic leaked out of a full nozzle
even when there is no pressure at all.

A proper model of pressure decay is quite difficult, and would improve only
marginally for such short pieces which are coasted.

What glitchiness are you referring to?

@Ghostkeeper are you sure the bug is introduced by that single parameter?
It might be the case that enabling coasting causes a time estimation bug to
become apparent for example.

CCS86, 2015-09-25:
@BagelOrb 

I reported it here: https://github.com/Ultimaker/Cura/issues/381

Ghostkeeper, 2015-09-26:
> @Ghostkeeper are you sure the bug is introduced by that single parameter? It might be the case that enabling coasting causes a time estimation bug to become apparent for example.

Enabling coasting while keeping all other settings the same cause the loading bar in Cura front-end to freeze and prevents me from saving the gcode or viewing the layers, so I assumed that CuraEngine freezes or at least stops sending progress updates and gcode back. I haven't tried to call CuraEngine without front-end though. Perhaps you can show me how that's done when you're back at work?

BagelOrb, 2015-09-30:
The percentages of the coasting speeds compared to the printing speed was taken to be a proportion, so the effective speeds were 100 times too high, causing an assertion fail, which somehow didn't result in the frontend reporting an engine crash.

BagelOrb, 2015-09-30:
solved in 15.10

closed 2015-09-30 by BagelOrb