Ultimaker/CuraEngine #250: [master + 15.10] Undeclared variable

thopiekar, 2015-09-28:
Just got notified about a failing log. It says that there is a undeclared variable.

Just check the log above:
https://launchpadlibrarian.net/219049939/buildlog_ubuntu-vivid-i386.cura-engine_15.01-0~201509281846~rev636~pkg9~ubuntu15.04.1_BUILDING.txt.gz

Ghostkeeper, 2015-09-28:
Thanks for the notification! Something seems to have gone gravely ill there. For the lazy debuggers, this is the relevant section of the log:

    /«PKGBUILDDIR»/src/FffGcodeWriter.cpp: In member function 'void cura::FffGcodeWriter::writeGCode(cura::SliceDataStorage&, cura::TimeKeeper&)':
    /«PKGBUILDDIR»/src/FffGcodeWriter.cpp:42:43: error: 'totalLayers' was not declared in this scope
         gcode.writeComment("LAYER_COUNT: %d", totalLayers);
                                               ^
    /usr/bin/cmake -E cmake_progress_report /«PKGBUILDDIR»/obj-i686-linux-gnu/CMakeFiles 6
    [ 18%] Building CXX object CMakeFiles/CuraEngine.dir/src/FffPolygonGenerator.cpp.o
    /usr/bin/i686-linux-gnu-g++    -std=c++11 -Wall -static-libstdc++ -I/«PKGBUILDDIR»/obj-i686-linux-gnu -I/«PKGBUILDDIR»/libs    -o CMakeFiles/CuraEngine.dir/src/FffPolygonGenerator.cpp.o -c /«PKGBUILDDIR»/src/FffPolygonGenerator.cpp
    /usr/bin/cmake -E cmake_progress_report /«PKGBUILDDIR»/obj-i686-linux-gnu/CMakeFiles 7
    [ 21%] Building CXX object CMakeFiles/CuraEngine.dir/src/FffProcessor.cpp.o
    /usr/bin/i686-linux-gnu-g++    -std=c++11 -Wall -static-libstdc++ -I/«PKGBUILDDIR»/obj-i686-linux-gnu -I/«PKGBUILDDIR»/libs    -o CMakeFiles/CuraEngine.dir/src/FffProcessor.cpp.o -c /«PKGBUILDDIR»/src/FffProcessor.cpp
    CMakeFiles/CuraEngine.dir/build.make:133: recipe for target 'CMakeFiles/CuraEngine.dir/src/FffGcodeWriter.cpp.o' failed
    make[3]: *** [CMakeFiles/CuraEngine.dir/src/FffGcodeWriter.cpp.o] Error 1
    make[3]: *** Waiting for unfinished jobs....
    make[3]: Leaving directory '/«PKGBUILDDIR»/obj-i686-linux-gnu'
    make[2]: *** [CMakeFiles/CuraEngine.dir/all] Error 2
    CMakeFiles/Makefile2:63: recipe for target 'CMakeFiles/CuraEngine.dir/all' failed
    make[2]: Leaving directory '/«PKGBUILDDIR»/obj-i686-linux-gnu'
    make[1]: *** [all] Error 2
    dh_auto_build: make -j4 returned exit code 2
    Makefile:140: recipe for target 'all' failed
    make[1]: Leaving directory '/«PKGBUILDDIR»/obj-i686-linux-gnu'
    debian/rules:7: recipe for target 'build-arch' failed
    make: *** [build-arch] Error 2
    dpkg-buildpackage: error: debian/rules build-arch gave error exit status 2
    ────────────────────────────────────────────────────────────────────────────────
    Build finished at 20150928-1849
    
    Finished
    ────────
    
    E: Build failure (dpkg-buildpackage died)

But I don't see how this ever could go wrong, since this is the relevant portion of code in branch 15.10:

    unsigned int total_layers = storage.meshes[0].layers.size();
    gcode.writeComment("LAYER_COUNT: %d", totalLayers);

And this is the relevant portion of code in the master branch:

    unsigned int total_layers = storage.meshes[0].layers.size();
    //gcode.writeComment("Layer count: %d", totalLayers);

    bool has_raft = getSettingAsPlatformAdhesion("adhesion_type") == EPlatformAdhesion::RAFT;
    if (has_raft)
    {
        processRaft(storage, total_layers);
    }

It's clearly defined there :confused:

thopiekar, 2015-09-28:
No problem. It could be important to debug the problem. Because the master
branch is not often in sync I made the launchpad build system merge all
commits which are new in 15.10 branch into master.
Of course this is not save, but needed for the Cura package because I
wanted to build the latest code from 15.10 but needed the fixes I sent to
master.

nophead, 2015-09-28:
total_layers vs totalLayers

BagelOrb, 2015-09-28:
So you cherry picked some commits from a different branch. You can expect there to be some merge conflicts. 

We don't have this problem in either branch. The problems are by your own doing and you should solve the problem by changing some code,  ie the variable name.

Ghostkeeper, 2015-09-28:
Oh good catch @nophead, totally glanced over that. Just broken in 15.10 then.

BagelOrb, 2015-09-29:
It seems this bug is present in the 15.10 branch itself - even without merging with the master branch.

git blame reveals possible culprits: @oliv3r and/or @nallath and/or me.

nallath, 2015-09-29:
If it was a bug created by me, it has been there for a loong time.

oliv3r, 2015-09-29:
Time to own up, I added that and I fixed it, running some tests to see if it is properly fixed.

note to self, don't blalantly un-comment code  :)

BagelOrb, 2015-09-29:
Now also cherry-picked to 15.10

closed 2015-09-29 by BagelOrb