Ultimaker/CuraEngine #251: 15.09.85: Support Disorder

CCS86, 2015-09-30:
I seeing some defects in the support structures. One column is very chaotic, and other areas have small patches or notches:

[image]

I'll post up my setting tomorrow. Cura locked up while processing layers, and won't restart.

BagelOrb, 2015-09-30:
This is in version 15.10, right?

I haven't seen any such behaviour yet.

Can you send me the model?

CCS86, 2015-09-30:
15.09.85

Here is the model: https://dl.dropboxusercontent.com/u/60958586/lampshade4.stl

and support settings:
[image]

Ghostkeeper, 2015-09-30:
Pretty lampshade, by the way. Would project some nice patterns on the wall. Also an excellent stress test for Cura and your feeders :stuck_out_tongue:

CCS86, 2015-10-01:
> wall. Also an excellent stress test for Cura and your feeders [image:

Hahaha, it's a nightmare of a print! I've already printed the base for it,
and have been putting off the shade. It's a super cool design (not mine),
but really not designed for printability!

BagelOrb, 2015-10-01:
Changed the title, since we wouldn't like Cura to have support for disorder (except the fuzy skin mode)
;)

BagelOrb, 2015-10-01:
Debugging is a nightmare! This model is too high poly! It takes forever to only once produce the gcode and my gcode viewers crash :S

BagelOrb, 2015-10-01:
The old faithful Cura could visualize the gcode, but I don't have the jagged column.

BagelOrb, 2015-10-21:
How the support works now:
Every layer new overhangs might introduce new polygons in the support, which means the polygons of the layer need a new 'wash' of simplifyPolygons and and polygon merging operations (like `polygons.offset(+amount).offset(-amount)`).

How this could be enhanced:
only apply those operations to polygons which could be affected by those operations; not to polygons which will end up as the same as in the previous (above) layer.

See how to do this efficiently: 
Perhaps have aabb's for all existing and all new polygons (offset them by the `+amount`) and only perform the operations on the new polygons and old polygons for which the aabb's intersected.

closed 2016-10-10 by BagelOrb