Ultimaker/CuraEngine #253: 15.09.85: Not filling small gaps

CCS86, 2015-10-03:
The gap filling seems very inconsistent remove & compensate for wall overlaps disabled:

[image]

CCS86, 2015-10-03:
It would be really great if Cura could identify these opportunities for an additional shell pass that is less than a nozzle width. The volumetric calculation could now be based on the width of the section, rather than the width of the nozzle.

BagelOrb, 2015-10-04:
If you remove the overlapping walls, they will be gone. It's no use to then compensate for overlap, since the overlaps are already gone.

Perhaps these two boolean settings should be merged into a single drop down menu..

It's nearly impossible to identify these cases automatically. That will be out of scope.

CCS86, 2015-10-04:
That sounds like a separate issue to me. I just mentioned those settings in case they have the capacity to avoid filling these areas. 

@Ghostkeeper  This seems more like a bug in the gap filling to me. At least for the gap near the through hole. There is clearly enough room for a nozzle width infill there.

In regards to the long skinny gaps: on some layers they are filled and others they are not. The model thickness is uniform.

CCS86, 2015-10-04:
Model: https://dl.dropboxusercontent.com/u/60958586/Controller.stl

BagelOrb, 2015-10-04:
Yes. Remove overlapping wall parts avoids filing those areas.

For normal slicing you would need twice the nozzle size for it to be filled. For gaps between one and two times the nozzle size you need to enable compensate wall overlaps. For gaps between zero and one you need to enable fill inter perimeter gaps.

For gaps exactly the nozzle size the algorithms might alternate between inter perimeter gaps and overlapping walls due to rounding errors.

BagelOrb, 2015-10-04:
Remove overlapping wall parts is a fast algorithm which is on the safe side (no overextrusion) of doing it 'wrong'. It removes wall parts on purpose.

CCS86, 2015-10-04:
@BagelOrb 

I haven't seen a "fill inter perimeter gaps" setting. Where can I find it?

CCS86, 2015-10-05:
[image]

CCS86, 2015-10-05:
Vs old faithful:

[image]

BagelOrb, 2015-10-05:
It's called 'fill gaps between walls' in the GUI...

BagelOrb, 2015-10-05:
The 'old faithful' prints walls at very sharp angles which is impossible in practice. There you will see a bit of over extrusion. The new Cura performs under extrusion instead of over extrusion, when you enable 'fill gaps between walls'.

closed 2015-10-05 by BagelOrb

CCS86, 2015-10-06:
Looking better in the .90 build. 

There is still a visualization issue:

[image]

[image]