Ultimaker/CuraEngine #254: 15.09.85: Duplicate infill on same layer

CCS86, 2015-10-05:
I first noticed this behavior in watching this portion get laid down (not sure the name you have given it):

[image]

I saw it repeat that area without changing layers. This obviously lead to over-extrusion. I thought maybe I had been mistaken, but then started a new print and had similar, but worse behavior. Basically, after finishing layer 1, instead of moving on, Cura laid down the exact same infill a second time at the same height. No new perimeter was laid down. This behavior continued and I had to abort the print.

Settings:

[image]

Gcode:

https://dl.dropboxusercontent.com/u/60958586/UMO_Controller.gcode

Ghostkeeper, 2015-10-05:
Oh, maybe this is related to the overextrusion we've been seeing in the bottom layers recently?

BagelOrb, 2015-10-05:
This issue is already solved. I don't recall if there was already a github issue, though.

The problem was pretty grave. For every next top/bottom skin area all previous skin areas would be reprinted.

Gladly it's now resolved.

closed 2015-10-05 by BagelOrb

BagelOrb, 2015-10-05:
@awhiemstra When are you going to build another beta?