Ultimaker/CuraEngine #256: 15.09.90: Combing not optimized

CCS86, 2015-10-06:
I have seen some very long combing moves lately, which seem to dance all around the part, even passing by the end point only to return. I am pretty sure that it happened in this gcode, in the bottom couple of layers, but I'm having trouble finding the exact lines. Are the combing moves G1 or G0?

https://dl.dropboxusercontent.com/u/60958586/UMO_Controller.gcode

BagelOrb, 2015-10-06:
They are G0.

CCS86, 2015-10-06:
I figured after watching the moves, they would be easily identifiable in the code, but haven't located them yet. I'll do my best to nail it down the next time I see it.

Have you noticed this behavior, where a combing sequence takes ~4 times longer than it needs to, because of extraneous moves?

BagelOrb, 2015-10-06:
Well, from layer 6 onward the part consists of 4 narrow walls. The algorithm computing the combing paths highly depends on some heuristics which might fail in this case. 

The algorithm works in two steps. First it finds which walls intersect with a straight path to the destination and just walks along the walls starting from and ending in the straight path. Then it looks for consecutive line pieces for which it can take a short cut by skipping the middle point without crossing the walls.

So if the first part of the algorithm decides to take the long way around the square hole, then every short cut would cross the walls surrounding the hole, so the algorithm would still decide to take the long way around.

I'm not sure whether this applies to your situation. Please try to find the bad combing moves, cause I haven't found any in your gcode.

BagelOrb, 2015-10-08:
This might be related: https://github.com/Ultimaker/CuraEngine/issues/259

CCS86, 2015-10-20:
Will we see a new beta build soon?

Ghostkeeper, 2015-10-21:
Today we have a new build scheduled, iirc. It includes a few improvements to combing paths.

BagelOrb, 2015-10-23:
We've made a lot of improvements to combing.

I suppose these bugfixes also solve your issue.

You can send your model so that we can make sure your model is sliced correctly.

closed 2015-10-23 by BagelOrb