Ultimaker/CuraEngine #257: Major change without major release bump

cederom, 2015-10-06:
I just wonder why major rework of code in Cura and CuraEngine is done in the middle of 15 not 16rc1? People that get 15 think they get stable release while recent 15 releases introduced major changes in the code and dramatic bugs amount due to code reorganization.. :-(

BagelOrb, 2015-10-06:
It's done in 15, because that's the year we live in - 2015.

closed 2015-10-06 by BagelOrb

cederom, 2015-10-06:
Damn my time machine failed again! :-P

daid, 2015-10-06:
And there is a "legacy" branch that is made just before all the re-work was merged. That can still be used if the old interfacing is required.

BagelOrb, 2015-10-07:
Please keep us updated on the status of your machine. We are very interested, might it work some day.