Ultimaker/CuraEngine #258: Better line optimization.

BagelOrb, 2015-10-07:
Due to the gready nature of the path planning algorithm, sometimes a small area is skipped untill the end at which time we need a long travel move.

It would be better if an area would be 'completed' before moving on to the next.

===================================
One way to do this is to keep track of past positions and favour paths which are closer to the average location of the past couple of paths. We could keep track of a moving average position to do that.

Then when deciding which path is to be next, if it's really close by the actual location, choose that path; otherwise choose a location close to the moving average location.

=====================================
Another algorithm would do something along the following lines.

Store the second best path and choose the second best of the past couple of iterations if we're getting too far away and it hasn't been printed yet.

CCS86, 2015-10-07:
I like it!

CCS86, 2015-10-07:
I wonder if this could be related to the issue I posted here: https://github.com/Ultimaker/CuraEngine/issues/256

If the engine left behind a bunch of tiny print segments, then zoomed around to fill them, it would look like spastic combing, and the gcode wouldn't look overtly different (just a bunch of alternating print and rapid moves).

BagelOrb, 2015-10-08:
Indeed this could be related, but I'm not sure. The combing code does seem buggy in other tests I did.

caseorum, 2020-01-16:
I'm thinking of building optimal path planning into CuraEngine, and I wanted to gather thoughts on the matter before I start tinkering, since the changes would be somewhat wide ranging. Is this the correct place to discuss this, or would a new issue be more appropriate?

Every time this is discussed either here or in Slic3r, and even in the docs, it's mentioned that this is the Traveling Salesman Problem. In fact, it's the Chinese Postman Problem, and its solution is very straightforward, but all the print paths of a layer would have to be unified into one graph, and currently they are generated and stored at many places in the code and optimized separately.

Optimal path planning could also replace combing and, for single-material layers, collision avoidance. The extruder can be confined entirely to the print paths except to jump to a disconnected part in order to print it.

Would such an attempt be welcome and worth discussing? Or better left to my own greenfield project somewhere?

BagelOrb, 2020-01-16:
Let's discus first. The problem might be more involved than you think.

I don't think it's the route inspection problem exactly, because the there are multiple possible travewl paths between extrusion paths and not all possible travel paths have to be visited. Confining the nozzle to the print paths sounds like something which has a cost on total travel time, which could undo the reduction in print time gained by ordering the paths 'optimally'.

The goal of combing and collision avoidance is to avoid crossing the boundary between printed part and outside at the cost of extra print time. An algorithm only geared toward finding the fastest route wouldn't take such print quality constraints into consideration.

The routing algorithm should take both print quality considerations and print time into account. Currently the print quality considerations consist of boundary conditions (e.g. that the number of times the outline is crossed by the nozzle is minimal and that each polygonal path only has a single start/stop extrusion point). However, I would also be interested in an algorithm in which these considerations are not hard requirements, but weighed goals which get left behind when the print time penalty incurred by the detour is too large.

Optimal path planning could definitely replace all current pathing algorithms if the optimality criterion also involves such quality considerations. It would be interesting to theorize about how such an algorithm would look like!

caseorum, 2020-01-16:
Thank you for your interest! This is very encouraging.

The extrusion paths are fixed, but they are undirected and the travel paths are arbitrary, and so we are free to construct a postman problem. The entire graph must be connected, so given any two components, add an acceptable edge connecting them, repeat until the graph is connected. The choice of "acceptable" can encode almost any condition (I can think of only one requiring special care, the order of multiple perimeter walls.):

For the provably shortest tour, connect the euclidean-closest pair of nodes that are in different components. No tour can be shorter, but this will mean traveling skin to skin, so...

For the skin to be printed in a single extrusion, connect it at any, but exactly one, node to its part's internal structure only, and connect parts by a (perhaps the euclidean-closest) pair of internal nodes. That is, edge nodes behind the skins. This means the perimeter will be crossed once going in before it is printed, and once after to leave. To enforce that it crosses at exactly the same point both times, add a travel node outside the part and only connect other parts to that node.

For multiple-extruder prints, components of different materials are only connected via a node or edge that changes extruders. As you know, only one such change is required per layer for two extruders.

The travel edges added between odd nodes to complete the postman tour (the t-join) are also arbitrary, so take the Dijkstra-shortest path to stay above the print path for oozing, take the euclidean-shortest path for time, comb just like now, etc...

Each tour is for a single layer, and can be computed entirely in parallel with the other layers. There is an optimal choice of layer-change points making for a globally-optimal path, but there's a little bit of a trade-off in parallelism. There are other sensible choices of change points as well.

You mentioned having weighted goals. That shouldn't be necessary, and definite choices in the graphs are better. The choices above can be made in any way, but in terms of changing the underlying problem: Postman problems are strange in that even the slightest change kicks you back out into something NP-hard. Optional edges, mixed graphs,........

caseorum, 2020-01-17:
I'm being a little sloppy here, so I'll revise a little, and I need to be a
little clearer. Things can move too fast when you're having fun.

The obviously unnecessary travel described in this issue and in so many
others is within a single physically connected part of the layer (or do you
say island?), and within such a part, the postman tour is optimal.
Connecting all the disjoint parts into a minimal tour is still the
traveling salesman, and what I've described above would be pairwise nearest
neighbor.

I think that's good enough in any case and there's no need to really solve
the TSP here, but even exhaustive search is quick for the small number of
disjoint parts printed in practice, and nearest neighbor can be used if the
number gets too large. Is that the kind of weighed decision you were
thinking of?

BagelOrb, 2020-01-17:
The thing is that computing the combing path is currently very computationally intensive, so we definitely wouldn't want to compute the combing paths between each two possible extrusion start/stop locations.

Another thing to keep in mind is that polygons can be started printing anywhere along the polygon, while polylines can only be started at either end.

I don't really understand your proposed algorithms yet, but it's also too early for that. Let me just inform you on the difficulties and when you are convinced you have a good algorithm explain it here.

My weighted solution idea was to sometimes stop extruding the walls if you would otherwise leave a small inter-perimeter gap behind which you would otherwise have to do a long combing move for. It might sometimes be better to just do the inter-perimeter gap while printing the outline. This idea is a bit dangerous, though; switching between different print features introduces extra defects.

Ghostkeeper, 2020-01-17:
There are a couple of constraints to the line optimization. Some of these constraints actually make the problem easier, but some of them make it harder:
* It should avoid crossing (outside) walls, meaning that it should probably stay within one part. This makes the possibility space smaller so it's easier to compute though. Or otherwise there needs to be a big penalty for crossing a wall.
* Modelling accelerations, lines in the same direction are much cheaper than lines that move in a different direction since you won't need to slow down to the Jerk speed.
* We need to keep printing structure per structure, because 1) changing from e.g. infill to outer wall induces a big flow change which is detrimental to print quality, and 2) for different prints you need certain parts to be printed in a different order. For instance printing the outer wall first makes prints more accurate, but printing the inner wall first makes overhangs better. For 1) you could introduce penalties again (BagelOrb's weighted system) but for 2) you'd need to keep printing each structure separately or else also give penalties for e.g. printing over air while no other lines are next to it yet, which will be very hard to do.

Also consider the gains that can be made in our current algorithm are very small. Typically the time spent on travels is 5-10% of the total printing time. If you can save 10% on the travel time, you're saving 0.5-1% of the total printing time, which is not a whole lot.

caseorum, 2020-01-17:
There is no print path that crosses the perimeter, so the postman tour of
the part never crosses it. Travel has to be added to move into and out of
the part, just like now.

I'm surprised combing is hard. The replacement on a graph is just Dijkstra
shortest paths, which is pretty easy, and has the same effect as combing
without the strings and oozing off the paths.

Acceleration is easily minimized, because the choice of the next unprinted
edge at every node when following the tour is arbitrary. It's an Euler tour.

Any outer walls that are printed whole are irrelevant to the optimization
graph, especially if explicitly ordered. They are already optimal. They
would be represented by a single node. Any other print polygons or
polylines are edges in the graph, their intersection points are nodes. The
culprits for internal complexity are things like infill and skin.

Any penalties or explicit direction in the graph can make the problem
NP-hard, but order can be caused by how the graph is constructed or split.
Besides the walls, what other order is really necessary?

The t-join algorithm also informs how best to generate printable paths.
Travel is caused by odd nodes alone, so it is minimized when odd nodes are
fewest and clustered together in pairs close together.

Faster is better, and travel is the only waste in the print. The best case
is also very attractive: if every layer is connected and contains no odd
nodes, a one-material print is done in one continuous filament from the
bottom all the way to the top without a single travel move.

caseorum, 2020-01-20:
#1184 is just the most recent example of something that can be vastly
improved. Literally almost all of that travel is unnecessary.

My current plan is to use the Boost graph and geometry libraries. All of
the algorithms required for every stage are very standard and present in
those libraries.

Initially, it's good enough to solve the problem for every part alone, with
all the print paths as they are currently generated. This already brings
the largest benefit to be had. This replaces combing entirely, and
minimizes travel within the printed piece, everything else is untouched.
TSP and collision avoidance are fine for now.

I'll start showing you what I mean in code form as time permits.

>

BagelOrb, 2020-01-20:
It would be better if you can explain it in text, because reading code is generally more hard than natural language. Currently I don't really understand any of what you've tried to explain. Perhaps you could make a couple of drawings and graphs relating print moves with graph nodes/edges and with your algorithm idea.

> I'm surprised combing is hard. The replacement on a graph is just Dijkstra
shortest paths

I don't understand this at all, but I fear that you don't understand what combing tries to do. It finds the shortest path between two places which takes detours in order to avoid collision with some polygons. I'm not sure how you would want to do this, but if you put all possible travels in a graph then the size of your graph immediately explodes, because all possible travels are exponential in the vertex count.

> Any outer walls that are printed whole are irrelevant to the optimization
graph

The fact that the outer wall goes first still leaves open the order between all the outer walls of a single part, which may be multiple of the part has holes.

Ghostkeeper, 2020-01-20:
> There is no print path that crosses the perimeter, so the postman tour of
the part never crosses it. Travel has to be added to move into and out of
the part, just like now.

Ah, I'm sorry for the confusion. I thought you meant to optimise these as well.

> Besides the walls, what other order is really necessary?

See the current printing order here: https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_paths.md#printing-order . Apart from where the draft shield and the surface mode are printed, I'd say that this order is necessary. In fact it might be better to print the Surface Mode lines together with the outer walls, now that I think about it again... Something for another day I guess. Like I said though, this should make the optimisation easier.

> The t-join algorithm also informs how best to generate printable paths.
Travel is caused by odd nodes alone, so it is minimized when odd nodes are
fewest and clustered together in pairs close together.

This makes me realise that you have a different view of what Cura's movements look like than BagelOrb or me. T-joins pretty much don't exist in Cura's output. A line almost never ends up inside another line. Only really when the infill and overlap is set to 0 (which is not the default) will it happen decently often in a print, and even then there will be rounding errors to consider.

> #1184 is just the most recent example of something that can be vastly
improved. Literally almost all of that travel is unnecessary.

That issue is caused by each of the squares being a separate polygon. So it has a bunch of separate squares that need to be drawn. Most of the walls overlap with the walls of adjacent squares. As far as I could see (picked layer 170) the travel moves there were completely optimal apart from that it needs to go back to the middle for the next layer. It can't know the optimal path there because the layers are calculated asynchronously, so it doesn't know where the previous layer ended to start the next layer in an optimal position.

> My current plan is to use the Boost graph and geometry libraries.

This can be nice to demo, to see how much time you'll save with this!

In the past we've been very reluctant to pull in Boost as a dependency though because of difficulties to build it.

> Initially, it's good enough to solve the problem for every part alone, with
all the print paths as they are currently generated. This already brings
the largest benefit to be had. This replaces combing entirely,

You will still need to perform combing, even if you stay within a part or stay outside of a part (in the case of support). This is because the parts are not necessarily convex. When you travel through the inside of the walls of a maze you'll still need to avoid the walls and thus solve the maze.

> > I'm surprised combing is hard. The replacement on a graph is just Dijkstra
shortest paths
> I don't understand this at all, but I fear that you don't understand what combing tries to do. It finds the shortest path between two places which takes detours in order to avoid collision with some [...]

They are quadratic, not exponential. The confusion from Caseorum came from that we called combing "expensive" while it's actually only linear (Dijkstra's). We called it expensive because it's being executed inside a loop that's already quadratic.

caseorum, 2020-01-20:
> It would be better if you can explain it in text
>
Sure, I'll put a summary at the bottom here.

> I fear that you don't understand what combing tries to do.
>
That's very possible. It seems to be jargon, and I've never heard it in any
other context. I was under the impression that "combing" meant finding the
shortest travel path while staying within the printed work. Avoiding the
printed work while traveling outside the work I have called "collision
avoidance." If I'm wrong about that, please correct me. I would be curious
to know where the word comes from.

> The fact that the outer wall goes first still leaves open the order
> between all the outer walls of a single part, which may be multiple of the
> part has holes.
>
Yes. Because walls are printed as one loop that starts and stops at the
same point, there's nothing to optimize in that path. There's no fork in
the road to consider. Each wall will be printed whenever the print head
arrives at that point.

Small summary:
Consider a single physically connected print layer, one part. We wish for
the head to travel along each extrusion, if possible only once.
Take each extrusion to be an edge of a graph, with weight equal to its
length, and every point where the filament physically connects to be a
node. This graph is undirected because we don't care what direction any
line is printed in, and connected because the part is physically connected.
A node that connects an even number of edges is called "even," and
analogously "odd."
A path that follows every edge of a connected graph exactly once and ends
where it started is called an Euler Tour. There exists an Euler Tour if and
only if every node is even. So if every node is even, we already have the
minimal printing path.
If the graph contains odd nodes, we can duplicate an edge on each odd node,
making it even, and keep doing so until all nodes are even. Duplicating the
edge would mean traveling the line once and printing it once. Which edges
do we duplicate if we want to add as little length as possible to the tour?
That is, how do we add the least travel? This is the Chinese Postman
Problem. In contrast to almost every other problem in route planning, it is
solved relatively easily, and doesn't reduce to the Traveling Salesman
Problem. If you want the intuition of why that is, notice that the question
has nothing to do with order, just finding a minimal set of edges.
So now that our graph has only even nodes, it has an Euler Tour. We follow
it by starting anywhere, and at every node, following any edge we haven't
followed before.

BagelOrb, 2020-01-20:
> I would be curious to know where the word comes from.

We used to have no collision avoidance, so the code for combing is reused for collision avoidance.
I think it's called combing because the nozzle follows the contour, just like a comb follows your head. The resulting path is then simplified by taking straight shortcuts if those shortcuts don't cross the contour.

> Because walls are printed as one loop that starts and stops at the
same point, there's nothing to optimize in that path.

What's to optimize in that path is the start location along the polygon.

> Small summary

I think I understand now. However, your algorithm would work only when the extrusion paths are connected, but in reality they are very much disconnected. The Chinese Postman problem doesn't work if there are multiple disconnected components. We must first connect those separate components, so that the whole graph is reachable. Finding those connections is exactly what's the difficult part. Don't underestimate how difficult it is to compute those travel moves.

BagelOrb, 2020-01-20:
> The confusion from Caseorum came from that we called combing "expensive" while it's actually only linear (Dijkstra's).

Combing is not at all linear at the moment. The initial combing path is constructed by intersecting the contour polygons with a line, which is linear. However, the path is then optimized by checking for each vertex along the path whether removing that vertex will create a shortcut which doesn't cross the contour. Checking whether a line segment crosses a polygon is O(n), so computing a single combing path is O(n^2).

caseorum, 2020-01-20:
> computing a single combing path is O(n^2).

And that's counting every single vertex. Dijkstra is linearithmic, and
only in the number of nodes. I think collision avoidance can also be
done in linear or less, but that's for another time. In any case,
printing time matters more than slicing time, within reason.

caseorum, 2020-01-20:
> Ah, I'm sorry for the confusion. I thought you meant to optimise these as
> well.
>
The largest benefit is in optimizing within each part, so I'll start there.
That's the Generalized TSP outside the parts, and those moves seem good
enough as they are. I haven't seen anyone complain about those.

> See the current printing order here
>
Travel time can still be absolutely minimal as long as that order is only
locally true. That is, a wall is printed after all infill that touches it,
and then gap-filled.

> T-joins pretty much don't exist in Cura's output. A line almost never ends
> up inside another line.
>
Sorry, "T-Join" isn't a t-junction, it's a jargon term for the solution set
to the CPP. It's the minimal set of edges to be doubled. That's my bad,
I'll stop using that.

A node is only an abstract graph representation of connection. You could
say it's where segments cross, or where the filament touches, or where two
vertices are sufficiently close together...

> the travel moves there were completely optimal apart from that it needs to
> go back to the middle for the next layer.
>
I'm counting the optimum as 21 "square edges" per layer given the
wall-is-one-path restriction. It would be 11 if walls were allowed to start
and stop at sharp corners. That's an interesting idea.

> It can't know the optimal path there because the layers are calculated
> asynchronously, so it doesn't know where the previous layer ended to start
> the next layer in an optimal position.
>
That's one of the great things about an Euler Tour. You can calculate it
entirely independently per layer, and start anywhere on every layer.

> This can be nice to demo, to see how much time you'll save with this!
>
Exactly!

> In the past we've been very reluctant to pull in Boost as a dependency
> though because of difficulties to build it.
>
The algorithms are extremely standard, and can be implemented by anyone,
but that's still quite the repeated effort and there's often a huge gap
between implementing and implementing well. Boost already does them well.
What sort of difficulties? Solving them might be less effort on my part.

> You will still need to perform combing
>
Really not. All moves are along print paths, and you always end up where
you started. Dijkstra doesn't care about the shape, and the graph has no
concept of convex, only connected. That's one reason to do some things
using graph theory, and others using geometry.

caseorum, 2020-01-20:
> I think I understand now. However, your algorithm would work only when the
> extrusion paths are connected, but in reality they are very much
> disconnected. The Chinese Postman problem doesn't work if [...]
I mean points of physical connection, or points within a certain distance.
We're constructing every path, so we know where things "touch," and even if
we didn't, a spacial index could tell us very efficiently. If you mean
physically disconnected parts, I've seen them called "islands," then you
are absolutely right about that. In fact, since there's a choice of start
and stop points for each part which I have elided over a little bit,
absolute optimal outside travel moves are the Generalized TSP. The current
code is doing just fine out there.

BagelOrb, 2020-01-20:
I think these travel moves wouldn't be producible by your algorithm:
[image]

3D models:
[image]
infill: 0%

[cross_step.zip](https://github.com/Ultimaker/CuraEngine/files/4087458/cross_step.zip)

caseorum, 2020-01-20:
On the right, correct. Those parts are disjoint, and would use the TSP
heuristic. On the left, there is no travel necessary if the inner wall is
allowed to be printed in more than one run, and therefore allowed to
connect. If not, then those yellow parts are also disjoint.

>

BagelOrb, 2020-01-20:
Walls are not allowed to be broken up, so then your algorithm would produce a travel which detours via one of the outer ends of the + instead of going straight,

caseorum, 2020-01-20:
> Walls are not allowed to be broken up, so then your algorithm would
> produce a travel which detours via one of the outer ends of the + instead
> of going straight,
>
Internal walls? I can't see why, but in that case, the parts are not
connected, so the result would exactly like in the picture. No algorithm
would go around that wall.

>

caseorum, 2020-01-20:
Your point is correct in principle, though. The euclidean distance can be
less than the graph distance. The euclidean, or any other travel path can
be added to the graph to complete the tour. Even the current combing paths
can be used. I like traveling on the print lines because it contains
oozing, is very easy to compute, and automatically stays within the work
without any extra effort. Infill is very connected, so in practice, that's
a good trade-off. What kind of edge connects the odd nodes is a matter of
taste.

>

BagelOrb, 2020-01-20:
I think we should let all of these ideas sink in a bit and maybe come up with a solution to all of these in a couple of days.

caseorum, 2020-01-21:
That's a good idea.

Can I ask a few questions of your expertise in the meantime? Don't feel
obligated, I'm just somewhat new to the details of 3D printing and haven't
found these things discussed.

You mentioned changes in flow rate being bad for quality. When are they
absolutely needed in one print, and what does that effect on quality look
like?

Why are internal walls treated any differently than infill? The outside I
get, but aren't the internal walls just structure? They could have many
z-seams and nobody would know or care.

BagelOrb, 2020-01-21:
When changing feedrate / flow rate it takes a while to adjust so the lines are too thin or too wide for a little while.

If you first print the inner walls then this inaccuracy can affect the outer wall as well when it's being printed. I guess that when the outer wall is printed first then the inaccuracy is propagated inward, but there are other reasons to print the inner walls first - to do with overhangs for example.

Ghostkeeper, 2020-01-24:
If printing the inner wall first, blobs on the inner wall are visible on the outside as well (albeit a bit blurred) since the outer wall gets pushed out by them. If printing the outer wall first, this effect is gone. But we print the inner wall first by default to improve overhang quality.

caseorum, 2020-01-25:
My entire point, to sum it up, is that path planning is a graph problem.
Just having all the print lines in a graph has immediate benefits, and
opens up so many options for the future. You've called all of this "my
algorithm" but in fact all of these are standard solutions to standard
problems that have been around for decades and any graph library can do
them:

1. Printing a part is the Chinese Postman. Find paths between all odd nodes
(straight lines, Dijkstra, or see the next paragraph), run Edmond's Blossom
algorithm. O(n^3) in the number of odd nodes.

2. Connecting parts with collision avoidance and "combing paths" for the
above are all Euclidean Shortest Paths. Find the visibility graph (The
library that's publicly available uses Lee's algorithm which is old and
runs in O(n^2logn) in vertex count.) and run Dijkstra.

3. Choosing the order of parts is the Travelling Salesman. Any
approximation will do, but Christofides might be good option at some point
because it reuses Blossom. Run Prim's algorithm, then solve the Chinese
Postman on the minimum spanning tree. O(n^3) in the number of parts.

That gives you almost the best, and often exactly the best possible print
time using already-available code, which is important given that those
5%-10% can mean several hours of time saved on large prints.

But consider slicing time. If either of those heavy library algorithms are
found to be too slow in testing, they can remain an option for people with
impressive computers, or even eventually be replaced by implementing newer
ones.

I've now looked through the literature to see what can actually be usefully
implemented in the future. Blossom can be approximated well and visibility
graphs can be done exactly, both in linear time or nearly linear. The
implementations might take some work, though.

caseorum, 2021-10-30:
Someone filed immediately to patent this public conversation: US20200230888A1 . So the patent troll industry is alive and well.

BagelOrb, 2021-10-31:
Are you sure you linked the right patent? That one seems to eb about bridging, while this thread is about the order in which paths are printed.

caseorum, 2021-11-01:
<details>
 <summary>(Off-topic discussion of that patent)</summary>

Yes, it describes exactly what I suggested. That application looks to patent, essentially, "add travel between disjoint paths, connect the odd nodes somehow, use for ordering in additive manufacturing." Bridge path means travel. It's not a good patent, being neither novel nor non-obvious IMO, but I was too quick to judge the applicant. It's from Oak Ridge National Laboratory, who have apparently been doing exactly as I suggested here as part of their BAAM project since at least 2017 in their ORNL Slicer 2. It's not necessarily malicious since, if the US is going to issue a bad patent on one's work, it's at least best to hold that bad patent oneself.

I was briefly distracted by that application when checking to see if anyone had already published code for this since...
</details>

I'm actually looking to implement something fun soon and thought of two options on this topic.

1. Graph-based route planning as above. The toolhead not leaving the extrusion path would be an improvement in print quality and maybe save a little time. I might be able to fit it in as a change to the path order optimizer and the way travel is added. This would make combing unneeded and ignores the TSP.
2. Cool new combing. Shortest paths around holes is such an old problem with good algorithms because everyone thought it would be useful decades ago, but none of them have been implemented because... it's not. I'm not sure anyone would notice the change unless combing is taking up a lot of slicing time, but an efficient, simple, and long-ignored algorithm has its appeal.

Would either fit with your internal roadmap?

BagelOrb, 2021-11-01:
I'm not sure what you're suggesting.

1. TSP is too computationally intense and would cause slicing to take too long for a marginal gain in print time. Travels are performed fast and don't contribute a very big amount to the total print time. I'm not sure what other graph based algorithms would do.
2. I'm not sure what cool new combing algorithm you are suggesting.

caseorum, 2021-11-02:
1. My original suggestion in this thread. Ignore the TSP, connect all the paths together into one graph in whatever way, like connecting closest vertices, treat the whole graph as the route inspection problem. Arachne paths make simple graphs, by the way.
2. The algorithms themselves aren't new. Study mostly stopped in the nineties. For this application I'd use Kapoor et al. "An Efficient Algorithm for Euclidean Shortest Paths
Among Polygonal Obstacles in the Plane" using Hertel and Mehlhorn's "Fast Triangulation of the Plane with Respect to Simple Polygons," the improved O(n+log s) one, as the underlying triangulation. That's a long-winded way of describing a very straightforward plane sweep algorithm.

BagelOrb, 2021-11-02:
Sounds like some good ideas!

I'm not sure how much the computation time would increase because of 1, so we'll have to see. I'd rather see idea 2 being worked on, to be honest.

Idea 2 could even decrease the computation time, seeing as the current implementation is O(n^2); it currently produces a non-optimized path first, which includes some of the boundary polygons, so is O(n) and then for each vertex on the path it checks if removing the vertex would cause the path to intersect with any of the boundary polygons, which is also O(n), so combines is O(n^2).

I would be very interested to see idea 2 being worked out!

Don't include too much in a single PR. Doing path optimization across multiple print features (walls, infill, skin, etc) at the same time would be off limits because it influences print quality. Just try to write an alternative for the Comb class which uses the same interface - or as much as possible.

caseorum, 2021-11-05:
I'm looking forward to (2) as well. The plan for (2) is to keep everything separate and generic, and glue in a replacement for the `LinePolygonsCrossings::comb` calls. I'll work on it as time permits. The change to existing logic is that it always fails if there isn't a path. Conditional crossings are np-hard, but there could be some good-enough workarounds to think about later. Or fall back to the current bug algorithm.

I'll also note for my future self or whomever is interested:
* The proposed algorithms of (2) aren't limited to two target points, but can be used to find obstacle-avoiding shortest paths between any number of target *polygons* after a single plane sweep pass.

* Arachne is so cool. Arachne's graph is closely related to the graph for (1). Repeating central odd beads as travel is the simplest solution, but would be suboptimal for e.g. single-bead infill. Any solution, after joining any separate loops together, makes arachne self-planning for any polygonal area.

Ghostkeeper, 2021-11-08:
One other thing to keep in mind is that we'd also have to maintain the code for years to come, so it should also not get too complex. If the optimisation changes a 50-line function into a bunch of classes and 4000 lines of messy code, that could also be a reason to reject the change even if it improves the travel time by 2%.

Traditional graph-based route planning is in this case quite difficult since the possible travels are from any line segment end point to any other endpoint. So you'll end up with basically a complete graph, which is quadratic in itself already. You could have a heuristic to reduce the number of edges in the graph to only nearby paths (e.g. cut any edges longer than 1/4 the bounding box unless they break up the graph into multiple disconnected components). But it'll stay a relatively expensive way to go about it.

But I'm indeed very interested in the second idea (polygon avoiding algorithms). There is definitely lots of room for improvements in performance there, if not in optimisation.

caseorum, 2021-11-11:
@Ghostkeeper we're talking about different graphs. I mean a graph of only physically connected extrusion paths. More or less the line graph of arachne's. Think of arachne's beads as edges and transitions as nodes. Or think of the points where infill intersects the innermost wall as nodes.

You're talking about the one currently in curaengine, the bag of line segments with a special index. That treats the problem as a TSP by implicitly including all possible travel, but the TSP is not very large or relevant here. I would argue it's irrelevant:

Planning a single, physically connected part into a single tour is not the TSP. Minimizing total travel between parts is, but it's a small instance. But travel is fast, and viscosity will only hold material in the hot end for so long, so I suggest instead minimizing the length of individual travels between parts. Think a minimum spanning tree of the parts. It's also a byproduct of the tools in (2).

I'll convince you one day, even if it's not until you see a layer actually printed this way.