Ultimaker/CuraEngine #259: Combing bug

BagelOrb, 2015-10-08:
Combing seems not to be optimized at the crossing between going straight and avoiding an object.

[image]

BagelOrb, 2015-10-08:
There is another combing bug which causes the 'avoid other objects' to **fail**.

[image]

BagelOrb, 2015-10-08:
The latter bug is probably caused by faulty line - polyline intersection checks. The 'optimized' travel move crossed the polygon exactly at the crossings.

BagelOrb, 2015-10-13:
Another bug, possibly related to the first.

slice -v -j ../Cura/resources/machines/ultimaker2.json -s travel_avoid_distance=.1  -o "output/test.gcode" -l "/home/tim/Documents/3D_models/combing/combing_evasion_Cura_out.stl"

[image]

[combing_evasion_Cura_out.stl.txt](https://github.com/Ultimaker/CuraEngine/files/12326/combing_evasion_Cura_out.stl.txt)

BagelOrb, 2015-10-13:
This is the model of the other problematic case, if I'm right.
[cube_cm_forrest.stl.txt](https://github.com/Ultimaker/CuraEngine/files/12375/cube_cm_forrest.stl.txt)

Ghostkeeper, 2015-10-15:
I fixed the original issue (the optimisation of combing paths). The issue was very deep! I could never reproduce the second issue. The third issue, with the move path right through the two circles, still persists.

BagelOrb, 2015-10-15:
You seem to have solved the second bug as well. I reproduced the bug, pulled your commits, and saw that the problem was solved.

Ghostkeeper, 2015-10-16:
Perhaps that was dfd3184 then.

BagelOrb, 2015-10-20:
Most probably!

Ghostkeeper, 2015-10-22:
Got the third combing bug fixed in c0d22dcf7bf8c8383961520b7fe3d8294ab4e1b5. I should've added a `fixes #259` to that commit message...

More combing bugs remain, but different ones than the ones reported here.

closed 2015-10-22 by Ghostkeeper