Ultimaker/CuraEngine #260: Assertion fail writing GCode

Ghostkeeper, 2015-10-09:
I'm loading [this](http://dulek.net/work/oe.stl) model into CuraEngine via the command line and am getting the following error:

    ruben@UM-MBA-187:~/Projects/CuraEngine$ ./build/CuraEngine slice -v -j ../Cura/resources/machines/ultimaker2.json -o output/test.gcode -l "/home/ruben/Models/oe.stl" -s infill_sparse_combine=3

    Cura_SteamEngine version DEV
    Copyright (C) 2014 David Braam

    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU Affero General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU Affero General Public License for more details.

    You should have received a copy of the GNU Affero General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
    Loading polygon setting machine_disallowed_areas not implemented...
    Unrecognized data type in JSON: machine_platform_offset has type array
    Loading polygon setting machine_head_polygon not implemented...
    Loading polygon setting machine_head_with_fans_polygon not implemented...
    Unrecognized data type in JSON: machine_head_with_fans_polygon has type array
    Unrecognized data type in JSON: machine_disallowed_areas has type array
    Unrecognized data type in JSON: machine_platform_offset has type array
    Loading /home/ruben/Models/oe.stl from disk...
    Unregistered setting mesh_position_z
    Unregistered setting mesh_position_y
    Unregistered setting mesh_position_x
    Unregistered setting center_object
    Loaded from disk in 0.020s
    Unregistered setting extruder_nr
     -g -e0 -s machine_heat_zone_length="16" -s machine_nozzle_expansion_angle="45" -s machine_nozzle_head_distance="3" -s machine_nozzle_tip_outer_diameter="1" -s machine_nozzle_size="0.4" -e0 -l "0" -s infill_sparse_combine="3"
    Unregistered setting support_extruder_nr
    Unregistered setting support_roof_extruder_nr
    Progress: start accomplished in 0.000s
    Starting slice...
    Slicing model...
    Layer count: 998
    Progress: slice accomplished in 0.416s
    Starting layerparts...
    Unregistered setting multiple_mesh_overlap
    Progress: layerparts accomplished in 0.208s
    Starting inset...
    Progress: inset accomplished in 2.348s
    Starting support...
    Progress: support accomplished in 1.207s
    Starting skin...
    Unregistered setting ooze_shield_enabled
    Progress: skin accomplished in 4.272s
    Starting export...
    Unregistered setting machine_nozzle_offset_y
    Unregistered setting machine_nozzle_offset_x
    Unregistered setting machine_extruder_start_code
    Unregistered setting machine_extruder_end_code
    Unregistered setting switch_extruder_retraction_amount
    Unregistered setting switch_extruder_retraction_speed
    Unregistered setting switch_extruder_prime_speed
    Unregistered setting speed_prime_tower
    Unregistered setting prime_tower_line_width
    Unregistered setting prime_tower_flow
    Unregistered setting prime_tower_size
    CuraEngine: /home/ruben/Projects/CuraEngine/src/gcodeExport.cpp:218: void cura::GCodeExport::writeMove(int, int, int, double, double): Assertion `(Point3(x,y,z) - currentPosition).vSize() < (int64_t((300) * 1000))' failed.
    Aborted (core dumped)

It looks like it wrote some huge move to a point that was very very far away. The model itself is approximately 400x400x100 units large.

daid, 2015-10-13:
Should not be an asserting, assertion is a programming error, and this is a user error (too large model)

BagelOrb, 2015-10-13:
The assert fails because the model is too large, but the assert doesn't check for the actual size of the model.

The assertion should not be compiled for the release, since programs might use CuraEngine for very large models. 

We could want the case of too large models to be caught and handled properly; you may create a feature request for better error handling.

As for now: the assertion fails for debug reasons - which is a good thing.

closed 2015-10-13 by BagelOrb